Oppzippy/OpenSCQ30 #88: "No devices found"

Mrnofish, 2023-05-16:
First! \o/

I'm trying to try OpenSCQ30 but it's not picking up my Q30 at all. The device is paired, appears to be working correctly, the OEM app works, but OpenSCQ30 keeps not finding anything.

Looking at my logcat I have multiple:

```
E CarrierPrivilegesTracker Unable to find uid for package com.oppzippy.openscq30 and user 10
E AppInfoDashboard com.oppzippy.openscaq30 not found for user: 10
W PackageConfigPersister App-specific configuration not found for packageName: com.oppzippy.openscq30 and UserId: 0
```

LOS 20 (Android 13). Shelter is installed, work profile stays mostly off.

Oppzippy, 2023-05-16:
The obvious explanation would be that my assumption is wrong that all Soundcore Q30s have mac addresses starting with AC:12:2F, since that's how I'm currently filtering the listed devices. Please let me know if that is not what yours starts with.

Mrnofish, 2023-05-16:
That seems to be the case. It's currently listing as E8:EE:CC.

Oppzippy, 2023-05-16:
Alright, thanks. I'll adjust how I filter the device list. I hadn't referenced the official apk until now, but it looks like they are just checking if the device names start with "Soundcore", so I'll probably go with that.

Mrnofish, 2023-05-16:
Can't wait for the next version, hopefully I can get rid of the horrible stock app!

PS: I'm not an expert but I'd probably reword that post just for the sake of peace of mind.

Oppzippy, 2023-05-16:
Fixed in 56b913dd1201863afc80d56611838c61ecf039bb. Let me know if v1.2.0 works for you.

Mrnofish, 2023-05-17:
Device recognized and everything seems to be working correctly so far. Thank you very much!

closed 2023-05-17 by Mrnofish