Oppzippy/OpenSCQ30 #89: Increasing downward slider extension

Mrnofish, 2023-05-16:
To tame the exuberant bass on the Q30, I'm currently using JamesDSP which is great but uses significant amounts of power.

This AutoEQ profile reins in most of the excesses however it applies almost -20 dB of attenuation.

Would it be possible to have an option for extending the slider strength that much down?

Mind, it's not generally regarded as a good idea to boost bands all that much (going too far introduces distortion), so the upwards extension could probably be limited to +6 as in the official app.

The total extension would then be about the same you'd have with a +12/-12 range, at the same time making the most of it by offering actual usable range.

What says you?

[image]

Oppzippy, 2023-05-16:
The -12 dB lower limit is a limit of the headphones (or at least the communication protocol) since the volume is represented as a single byte, with 120 being 0 dB, 0 being -12 dB, and 240 being +12 dB.

I figured it's "probably fine" to go above the official app's +6 dB as long as you don't turn the volume up too much, but maybe I should have +6 dB be the maximum by default with an option to enable the extended upper limit. I'll think about it.

I hadn't considered this until now, but I wonder if 255 would be -12.1 dB, 13.5 dB, or what. I'll look into that as well, but even if it works, it would only get us an extra 1.5 dB.

Mrnofish, 2023-05-17:
Found'em!

https://fccid.io/2AOKB-A3028/Internal-Photos/Internal-photos-4855127

Oppzippy, 2023-05-17:
> You mean damage the headphones..? Distortion doesn't generally affect hardware AFAIK, it just destroys sound, but the US is an infamously litigious country...

My concern was what if it doesn't clip and instead gives the speakers more power than they're intended to handle? For example, if they decrease the volume in software before EQ enough to ensure that +12 dB won't clip, and then set the gain with that lowered volume plus 6 dB in mind. I'll probably just leave it as is for now since causing physical damage has enough unlikely conditions that I doubt it will occur.

I'm not really concerned about getting sued or anything if something goes wrong, mostly because this is just a small project with not many users, but also since the GPL includes [this part](https://github.com/Oppzippy/OpenSCQ30/blob/master/LICENSE.txt#L589-L598). My concern was more just whether or not it's possible to misuse the software by mistake and damage one's device.

> Have you already tried out figuring out what's inside the Q30, I assume it must be built from off-the-shelf components ? A datasheet could go a long way and I seem to recall the FCC has pictures of the innards of every approved product on their site.

Interesting, didn't know about that FCC site. It seems basically everything is done by the BES2300, which I haven't found much information about.

Mrnofish, 2023-05-17:
I did some more searching after my last post, the BES2300 is used in a number of devices, including the Pine64 buds IIRC.

You've probably found the same datasheet I did, that's mostly hardware specs and not much of interest ?

Ultimately the problem is possibly that, no matter what hardware runs the show, the app has to speak to the (proprietary) firmware.

Oppzippy, 2023-05-18:
> I hadn't considered this until now, but I wonder if 255 would be -12.1 dB, 13.5 dB, or what. I'll look into that as well, but even if it works, it would only get us an extra 1.5 dB.
I tried out it and 255 seems to be +13.5 dB, so no luck on getting increased downward range. I'm not sure there's much value in having a lopsided EQ that goes more up than down when we already have a total range of 24 dB, which should be plenty for anyone not doing subtractive EQ only.

Mrnofish, 2023-05-18:
Not the only decision I'd call odd in a product that is surprisingly decent, in spite of everything!

After having used OpenSCQ30 a little... congratulations are in order for a pretty nifty app, with extra gratitude for letting me uninstall that bloated, only partly useful thing that is the stock app. Having CLI and GUI apps for computers is an especially nice touch: most manufacturers don't even care.

BTW and please don't take this the wrong way, have you considered adding support for some other Anker products at some point?

Oppzippy, 2023-05-18:
> have you considered adding support for some other Anker products at some point?

I have thought of it but I haven't made a decision one way or the other yet. The original goal of the project is mostly accomplished at this point, so it's really just deciding I want to expand the scope or move on to something else. So, my answer for now is maybe.

Oppzippy, 2023-09-09:
v1.6.0 adds experimental support for [many more device](https://github.com/Oppzippy/OpenSCQ30/blob/de6e96bfec765cc9f772d27fa5b087a31cb3dc84/README.md#supported-devices).

closed 2023-09-09 by Oppzippy