Oppzippy/OpenSCQ30 #95: Improving disconnection handling on Desktop?

Mrnofish, 2023-06-21:
Whenever the Q30 is disconnected and reconnected, even without disabling BT on the host, it seems like there is no way around closing and re-launching the program.

The settings window remains open but any changes to the controls simply result in an error message, whereas if I recall correctly, at least the mobile version drops back to the device selection screen, or perhaps the app is stopped and restarted by the OS ?

Oppzippy, 2023-07-04:
Fixed on all platforms in v1.4.0. The desktop client currently doesn't register that it's disconnected until you perform an action that fails due to the lack of an open connection, so I'll leave this open until I make it consistent with Android, which goes back to device selection immediately upon device disconnection.

Oppzippy, 2023-07-11:
Desktop gui is fixed in v1.5.0.

closed 2023-07-11 by Oppzippy