Oppzippy/OpenSCQ30 #99: Rename Project

Oppzippy, 2023-09-09:
Now that more devices than just the Life Q30 are supported, the project should probably be renamed. I'm unsure if a name containing "Soundcore" would violate their trademark, so I'd rather avoid anything that is too direct.

SpacingBat3, 2023-09-11:
@Oppzippy what will happen to Flatpak packages – do you also plan changing the package ID? I also maintain the AUR package and I'm not sure if I should also submit a deletion request to open it on a new repo once there will be a new name or just rename the individual packages but leave old `$pkgbase` as-is. And since GIT repositories in AUR can't be really renamed and aren't usually *trully* deleted (everyone could still *undelete* the package AFAIK just by `git pull && git push`), I'm not really sure myself which approach I should take.

Oppzippy, 2023-09-11:
I haven't fully thought this through yet, so I don't really have a plan or anything. I wasn't aware that anyone was packaging the application besides myself, so that does complicate things. I may just leave the name as then if changing it would cause trouble for myself with the flatpak and for other package maintainers.

SunDevil311, 2023-12-19:
Why don't you just get rid of the 30 in the name? Thus, instead of OpenSCQ30, just change it to OpenSCQ.

If you want, you could try to jazz it up a little bit... maybe OpenSC.Q or OpenSC-Q? Just an idea that came to mind.

KR4U7H3R, 2024-01-05:
I think "OpenSC" is fine as I liked the original name. Of course someone can come up with a fancy new name

Oppzippy, 2024-08-31:
I decided to just leave the name as is since a FOSS project with the name [OpenSC](https://en.wikipedia.org/wiki/OpenSC) already exists, and I'd rather keep the name google-able. The current name is not good, but it's unique, which I consider a significant benefit.

closed 2024-08-31 by Oppzippy