Oppzippy/OpenSCQ30 #100: Custom settings exposed through Presets but not under General on Q30

Mrnofish, 2023-09-12:
When using 1.6.0 with the Soundcore Q30, the switch for Transparency and ANC tunable are exposed through the Presets UI, but Live settings only allow selecting stock options.

Since the additional controls are missing, activating a Preset that uses Custom ANC also causes all the options in the radio switch under General to become deselected.

Either Q30 supports those modes and General should display the relevant settings, or the extra options should be hidden under Presets.

closed 2023-10-01 by Mrnofish

Oppzippy, 2023-10-01:
Oh, I forgot to close the issue. As you found out, fixed in v1.7.0.

Mrnofish, 2023-10-01:
Thanks for taking care of it.

Any news on Presets making their way to the Desktop app?

Oppzippy, 2023-10-02:
I wasn't aware that presets were wanted on platforms other than Android, so no news yet other than that it'll happen.

Oppzippy, 2023-10-22:
> Any news on Presets making their way to the Desktop app?
Added in v1.8.0