Oppzippy/OpenSCQ30 #101: support Space A40

sybenx, 2023-10-20:
Add support for Space A40 earphones. 
https://us.soundcore.com/products/space-a40-a3936011

RodoMa92, 2024-03-11:
Hci logs with the app [here](https://github.com/gmallios/SoundcoreManager/issues/19#issuecomment-1988910089) for now.

Oppzippy, 2024-12-01:
The Space A40 should be working as of v1.14.1. Let me know if there are any issues.

RodoMa92, 2024-12-01:
Doesn't really seems to detect them:
```
❯ ./openscq30_cli-linux-x86_64 get ambient-sound-mode
No device found.
```

RodoMa92, 2024-12-01:
From a quick spelunking in the code, this assumption might be wrong:
https://github.com/Oppzippy/OpenSCQ30/blob/0da38a7a90245b7805d8033001aaf9b4a23d3d07/lib/src/device_utils.rs#L11

```
	Name: Soundcore Space A40
	Alias: Soundcore Space A40
	Class: 0x00248404 (2393092)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Cont.. (0000110f-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Client   (0000112e-0000-1000-8000-00805f9b34fb)
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Audio             (00001203-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (00007034-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (0cf12d31-fac3-4553-bd80-d6832e7b3936)
```

The closest looking one on mine seems to be the second one from the bottom, but it starts with 7034, and it is not in range of 7777-8888 if I'm reading the code correctly.

Oppzippy, 2024-12-01:
I actually forgot when I was looking at the logs you posted that the MAC address was something unexpected, so that would be part of the problem too. I'm filtering for devices with mac addresses starting with AC:12:2F or E8;EE:CC, but yours starts with A4:77:58, right? Looks like that range is owned by Ningbo FreeWings Technologies Co.,Ltd, so I'll add their ranges to the list.

Regarding the `bluetoothctl info` output, it looks like it only includes services, but not GATT characteristics. Please do the following:
1. `bluetoothctl`
2. `scan on` and then `scan off` (if needed)
4. `connect YOUR_DEVICES_MAC_ADDRESS` - this is needed even if already connected, since it will cause GATT services to be discovered, and that does not occur automatically.
5. Post the output of this command:
```bash
bluetoothctl gatt.list-attributes | grep -E "/org/bluez/.*/dev_(AC_12_2F|E8_EE_CC|A4_77_58|A0_E9_DB)_" | xargs -I {} bluetoothctl gatt.attribute-info "{}"
```

RodoMa92, 2024-12-01:
> I actually forgot when I was looking at the logs you posted that the MAC address was something unexpected, so that would be part of the problem too. I'm filtering for devices with mac addresses [...]

Correct, that is my manufacturer Mac address reported by my buds. I'll update you with the correct Bluetooth params shortly, I'm not that versed in Bluetooth land myself :)

RodoMa92, 2024-12-02:
Initially I was confused, since list-attributes didn't return anything at all. Then I remembered that attributes are a specific feature of bluetooth LE, not classic, which was the more probable endpoint that I was connecting to by default. After running scan LE, I found another chatty bluetooth LE endpoint that starts with "E4_9E_58'', and after modifying your grep command by adding that filter like
```
bluetoothctl gatt.list-attributes | grep -E "/org/bluez/.*/dev_(AC_12_2F|E8_EE_CC|A4_77_58|A0_E9_DB|E4_9E_58)_" | xargs -I {} bluetoothctl gatt.attribute-info "{}"
```
(which, interestingly enough, doesn't seem to map to anything at all as a vendor mac address on bluetooth website),  I got the following output:
```
Service - Unknown
        UUID: Unknown                   (0149f5da-0000-1000-8000-00805f9b34fb)
        Primary: yes
Characteristic - Unknown
        UUID: Unknown                   (00008888-0000-1000-8000-00805f9b34fb)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0012
        Notifying: no
        Flags: notify
        MTU: 0x00dc (220)
Descriptor - Client Characteristic Configuration
        UUID: Client Characteristic C.. (00002902-0000-1000-8000-00805f9b34fb)
        Characteristic: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0012/char0015
Characteristic - Unknown
        UUID: Unknown                   (00007777-0000-1000-8000-00805f9b34fb)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0012
        Flags: read
        Flags: write-without-response
        Flags: write
        MTU: 0x00dc (220)
Service - Vendor specific
        UUID: Vendor specific           (66666666-6666-6666-6666-666666666666)
        Primary: yes
Characteristic - Vendor specific
        UUID: Vendor specific           (77777777-7777-7777-7777-777777777777)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service000e
        Notifying: no
        Flags: write-without-response
        Flags: notify
        MTU: 0x00dc (220)
Descriptor - Client Characteristic Configuration
        UUID: Client Characteristic C.. (00002902-0000-1000-8000-00805f9b34fb)
        Characteristic: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service000e/char000f
Service - Unknown
        UUID: Unknown                   (00007033-0000-1000-8000-00805f9b34fb)
        Primary: yes
Characteristic - Public Browse Root
        UUID: Public Browse Root        (00001002-0000-1000-8000-00805f9b34fb)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0008
        Notifying: no
        Flags: notify
        MTU: 0x00dc (220)
Descriptor - Client Characteristic Configuration
        UUID: Client Characteristic C.. (00002902-0000-1000-8000-00805f9b34fb)
        Characteristic: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0008/char000b
Characteristic - Browse Group Descriptor Service Class
        UUID: Browse Group Descriptor.. (00001001-0000-1000-8000-00805f9b34fb)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0008
        Flags: write-without-response
        MTU: 0x00dc (220)
Service - Generic Attribute Profile
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        Primary: yes
Characteristic - Service Changed
        UUID: Service Changed           (00002a05-0000-1000-8000-00805f9b34fb)
        Service: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0004
        Notifying: no
        Flags: indicate
        MTU: 0x00dc (220)
Descriptor - Client Characteristic Configuration
        UUID: Client Characteristic C.. (00002902-0000-1000-8000-00805f9b34fb)
        Characteristic: /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0004/char0005
```

Currently your app now sees my buds but the app can't find this specific service:
```
  2024-12-02T08:00:09.630760Z ERROR  select device

Caused by:
    service `011cf5da-0000-1000-8000-00805f9b34fb` not found: None
```

The closest one for me looking at it is the first one that starts with 0149f5da. I can also see now the two characteristics at 7777 and 8888, so it seems that I'm looking at the right thing RN.

RodoMa92, 2024-12-02:
I've changed the service endpoint to 0149.. and added the mac address to the filter, it does seems to try and negotiate further now, but it still explodes in the end with this:

```
  2024-12-02T08:23:31.361569Z TRACE  PropertiesChanged for /org/bluez/hci0/dev_E4_9E_58_XX_XX_XX/service0012/char0015: PropertiesPropertiesChanged { interface_name: "org.bluez.GattCharacteristic1", changed_properties: {"Value": Variant([9, 255, 0, 0, 1, 1, 1, 153, 0, 1, 1, 5, 3, 1, 1, 48, 52, 46, 49, 57, 48, 52, 46, 49, 57, 51, 57, 51, 54, 97, 52, 55, 55, 53, 56, 51, 100, 100, 97, 57, 101, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 84])}, invalidated_properties: [] }
    at /home/marco/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bluez-async-0.7.2/src/events.rs:165
    in btleplug_connection inbound_packets_channel reader

  2024-12-02T08:23:31.361631Z TRACE  event: "btleplug notification", data: ValueNotification { uuid: 00008888-0000-1000-8000-00805f9b34fb, value: [9, 255, 0, 0, 1, 1, 1, 153, 0, 1, 1, 5, 3, 1, 1, 48, 52, 46, 49, 57, 48, 52, 46, 49, 57, 51, 57, 51, 54, 97, 52, 55, 55, 53, 56, 51, 100, 100, 97, 57, 101, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 84] }
    at lib/src/soundcore_device/connection/btleplug/btleplug_connection.rs:206
    in btleplug_connection inbound_packets_channel reader

  2024-12-02T08:23:31.361697Z TRACE  PropertiesChanged for /org/bluez/hci0/dev_E4_9E_58_DA_3D_77/service0012/char0015: PropertiesPropertiesChanged { interface_name: "org.bluez.GattCharacteristic1", changed_properties: {"Value": Variant([9, 255, 0, 0, 1, 1, 1, 153, 0, 1, 1, 5, 3, 1, 1, 48, 52, 46, 49, 57, 48, 52, 46, 49, 57, 51, 57, 51, 54, 97, 52, 55, 55, 53, 56, 51, 100, 100, 97, 57, 101, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 84])}, invalidated_properties: [] }
    at /home/marco/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bluez-async-0.7.2/src/events.rs:165

  2024-12-02T08:23:31.362004Z ERROR  select device

Caused by:
    0: TryIntoInboundPacketError { message: "Error(VerboseError { errors: [([255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], Nom(MapOpt)), ([255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], Context(\"custom button model\")), ([1, 1, 5, 3, 1, 1, 48, 52, 46, 49, 57, 48, 52, 46, 49, 57, 51, 57, 51, 54, 97, 52, 55, 55, 53, 56, 51, 100, 100, 97, 57, 101, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], Context(\"a3945 state update packet\")), ([1, 1, 5, 3, 1, 1, 48, 52, 46, 49, 57, 48, 52, 46, 49, 57, 51, 57, 51, 54, 97, 52, 55, 55, 53, 56, 51, 100, 100, 97, 57, 101, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], Nom(Alt))] })" }
[... cut ...]  120, 120, 120, 120, 120, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 14, 0, 17, 0, 0, 17, 99, 17, 102, 17, 68, 17, 73, 7, 2, 48, 0, 1, 0, 0, 0, 8, 49, 0, 1, 1, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], Nom(Alt))] })
    at gui/src/main.rs:288
```

Oppzippy, 2024-12-02:
Huh, very strange that it has a different mac address for bluetooth and BLE. I added that mac address: 888e4c52bb04348892306a863d533be6b11536ad and fixed the `TryIntoInboundPacketError`: c9ec7cdf3b56e16c3ae7df14b828e1abce8cd102. Hopefully that gets it working*.

\* `./openscq30_cli-linux-x86_64 get ambient-sound-mode` won't actually work right now since I never added support for the space a40's sound modes to the CLI. Equalizer stuff should work, though. I'll implement the missing CLI features within the next few days and then do another release.

RodoMa92, 2024-12-02:
Oh, wow, thanks! Yep, now it works 100%, but you might still need an additional tweak on connect:

This device is weird, it does present two bluetooth address endpoint, to get the other address you firstly need to connect to the classic one, and then manually connect to the other endpoint. Then just using the second one makes the app work correctly 100%. There are still a couple of unhandled packets that pop up in the debug log when using the GUI, but this can be fixed later for sure.

The BLE address once paired to the classic one need to be manually connected before the app will show the second endpoint and connect to it, otherwise it will only show the primary and just spit out "device BLE service not found". I would probably just hide the 4A:77:58 if the app will find another address on E4:9E:58 with the same name, so at least it will be less confusing to the user (since I can't be sure if ALL soundcore devices have the same behavior).

Weird stuff, but from the GUI alone it does seem to completely expose every control that their app from Android do, amazing work on this! I can at least not use closed source stuff (besides for firmware updates I guess).

Again, thanks a lot for this! Now I can use this app to integrate this functionality in a Decky Plugin for my Deck and be able to control them directly from game mode :)

RodoMa92, 2024-12-02:
Tried to implement the buds in the case one, but I'm not sure that the debug is printing the complete command, and I'm a noob in Rust. Besides, I've found another handler for [9, 255, 0, 0, 1, 1, 4], so are they reusing the same commands with a different packet length? So nice of them /s

I'll probably wait for @Oppzippy to fix these, I need to start from something easier to understand Rust I guess :P

Tater-K7S5A, 2024-12-06:
I was just trying to get this working with my Space A40 buds this morning.  Just posting to encourage you.  Your work is appreciated.  Thank you.

Oppzippy, 2024-12-09:
@RodoMa92 Missing packet handlers are fixed in 5d11b3cba4187a9674cbb5f73946864c3d2e9869. They're all implemented already, just mistakenly not enabled for the space a40. `[9, 255, 0, 0, 1, 1, 5]` is implemented [here](https://github.com/Oppzippy/OpenSCQ30/blob/5d11b3cba4187a9674cbb5f73946864c3d2e9869/lib/src/devices/standard/packets/inbound/firmware_version_update_packet.rs), and `[9, 255, 0, 0, 1, 1, 4]` is implemented [here](https://github.com/Oppzippy/OpenSCQ30/blob/5d11b3cba4187a9674cbb5f73946864c3d2e9869/lib/src/devices/standard/packets/inbound/battery_charging_update_packet.rs) if you're curious.

> `WARN  A3936: unknown button action 0x49`

The left four bits is the action when TWS (true wireless, both earbuds together) is disabled, and the right four bits are the action when TWS is enabled. At least, that's my understanding of it. I'm not sure what causes them to not both be the same value. As shown [here](https://github.com/Oppzippy/OpenSCQ30/blob/5d11b3cba4187a9674cbb5f73946864c3d2e9869/lib/src/devices/standard/structures/custom_button_model.rs#L135-L144), known actions only go up to 6, so the 4 action for when TWS is disabled is known. It's the 9 that I don't know the meaning of. I added more context to the error message so you'll know which button has the problematic action.

> One last thing: rarely it does seem to stop responding to packets and on connect the UI will show a timeout, in this case just reconnecting to the bluetooth LE endpoint seems to have fixed it. Maybe another fallback to implement in case.

Yeah, I've been meaning to have it attempt to reconnect before bailing completely.

> I've also found that LDAC is already handled in the code, so I guess that the GUI is simply not been updated yet to expose that functionality?

Only parsing the packet that states whether or not it's enabled is implemented, so there is still more work to be done here.

Also, most missing cli features are added. Button actions are the only thing missing now.

I'll do a release either later later today or tomorrow.

RodoMa92, 2024-12-10:
Oh, another detail I just noticed.

New version fixed almost all packet parsing issues, just a couple of things left now:

I realized that the button mappings are not correct with the app, I'll attach a couple of pictures with the features name here (single tap seems to be limited as features compared with longer tap and hold) so you can correctly override them (or I'll do this myself and then push it here, whatever comes first).
[image]
[image]
[image]

Second, another packet is not handled, this is sent by the device when it is toggled in "game mode" (aka low latency). Here is the output of the app for it:

```
  2024-12-10T10:03:16.717826Z  WARN  no packet handler found for packet: Packet { command: Command([9, 255, 0, 0, 1, 1, 17]), body: [1] }
    at lib/src/soundcore_device/device/soundcore_device.rs:118
```
One is on (low latency active), zero is off (low latency inactive). The app can toggle this on from the gui, but I'm not sure on how to simulate that on my end.

Third: I still require to connect the BLE endpoint manually each time, it will not do this automatically, which is kinda annoying. If you can implement this directly within the app then this would be perfect at that point.

I still get the button on connect, but I do not have any way to toggle it manually. Single/Long/Double tapping both earbuds doesn't trigger that event at all. It might just be a firmware bug/random event triggered on connect from the firmware, I have no clue.

Other than that, I can't find any other problems that comes to mind. Again, thanks a lot for your time on this issue :)

RodoMa92, 2024-12-10:
I can confirm that adding a GameMode=9 to that structure makes it populate correctly, @Oppzippy . Not sure why the single clicks are still not detected (even if I tried adding two bogues entries for 7 and 8 in that structure, they are still seen as disabled).

RodoMa92, 2024-12-10:
Just tested the CLI and sadly it has the same issue as the GUI on disconnect, I need to connect and disconnect it manually each time to get it to be able to interact with the buds, otherwise they just fail with
```
Error: "Error fetching device: timed out: resending packet until ack received"
```

Oppzippy, 2025-01-08:
> I realized that the button mappings are not correct with the app, I'll attach a couple of pictures with the features name here (single tap seems to be limited as features compared with longer tap [...]

It turned out that the way I thought the button related stuff works is different from reality (for all devices), so I adjusted my implementation in v1.18.0. I'm not completely confident I got it right for the Space A40 though, so please let me know if there are still issues.

Single press having fewer options available seems to be a limitation of the soundcore app. I tried setting actions that the soundcore app does not allow for single press, and they seem to work fine.

> Third: I still require to connect the BLE endpoint manually each time, it will not do this automatically, which is kinda annoying. If you can implement this directly within the app then this would be perfect at that point.

This should no longer be an issue, since as of v1.18.0, bluetooth classic is used rather than bluetooth low energy.

RodoMa92, 2025-02-03:
Sorry for the delay @Oppzippy, these are my results:

- Connection now is fully fixed, I do not need to manually connect to the BLE endpoint, the app is automatically handling it in the background, so that is working :)
- Button mapping in the UI is now correct, but the app still times out on setting the parameter in the UI, regardless of the setting I choose. If I change them from the SoundCore app tho, the updated value is shown correctly in the UI from a quick test.
Besides the little problem above, once the missing packet handler are implemented correctly this app will be at feature parity with the SC closed source (besides firmware updates, but it's doable to just download it and then remove it once the firmware updates are applied IMO).

Thanks a lot again for this app :)