Oppzippy/OpenSCQ30 #102: QC35 verified working

SunDevil311, 2023-10-22:
Tested whether Soundcore QC35 headphones functioned properly with OpenSCQ30 app. A few issues noted under "Actual behaviour." 
Just performed a quick connectivity and music quality test, hope to have additional updates later on.

### Your environment
* Samsung Galaxy S20 5G
* Samsung OneUI 5.1 / Android 13
* Android security patch level: Oct. 1, 2023
* OpenSCQ30 1.8.0 downloaded from IzzyOnDroid via Droid-ify / F-Droid

### Steps to reproduce
Turn on paired Soundcore QC35 Bluetooth headphones, open app.
Select "Soundcore QC35" device, listed by name and Bluetooth address

### Expected behaviour
Expected app to function without issue on the pre-paired QC35 headphones.

### Actual behaviour
App functioned more or less as expected:
* Occasional latency making the Bluetooth connection is the most obvious thing I noticed.
* Sometimes it's quicker just to hit "back" and select the device again, after which it will often open settings immediately.
* Noticed some audio distortion during phone call (could easily have been the signal), as well as when switching between recent apps. I haven't been able to reproduce this last issue, will keep testing.

~ SunDevil311!

[Screenshot_20231022_141334_OpenSCQ30.png](https://github.com/Oppzippy/OpenSCQ30/assets/10506001/420c4782-db94-4d73-bd9d-da0f33b77906)

SunDevil311, 2023-11-02:
Now that I've spent a little more time with the app, I wanted to provide an update on the potential issues I had raised.

* Latency: still noticeable when the app is first launched, but the retry/timeout code added in 1.8.1 definitely help.
* I'm wondering if this is an actual app issue or some odd "warm-up" behavior with Bluetooth. Unclear what's causing the initial latency, but it's minor IMO. Maybe adding some sort of delay to the initial connection could help?
* Quick Profiles are working just fine. I initially misunderstood what that was supposed to do. I really like this feature, as it makes it easier to switch to "Normal" mode for phone calls. For some reason, people can't hear me well when I have noise cancelling on.

Overall, great app! Super simple to use and the EQ settings are much improved.

May give the Windows app a spin as well when I get a chance.

Oppzippy, 2024-01-03:
Thanks for all of the information. I marked the Q35 as working in the readme.

I doubt the distortion is related to the app, since the bluetooth LE service for controlling settings is completely separate from the audio. That's not to say it's impossible that there is some weird interaction though, so do let me know if there seems to be any pattern to it.

Not sure how much latency you're having for the initial connection, but since the timeout is 4 seconds (as of v1.8.1), it must not be longer than that. So, I'm going to leave that as is for now, since it's probably not worth the effort of diagnosing the issue and figuring out if it's even solvable.

closed 2024-01-03 by Oppzippy

SunDevil311, 2024-01-05:
@Oppzippy My bad, I could have easily made that change to the markdown file. Noted for the future.

Yeah, I never experienced any further distortion. Either I was mistaken or something else was going on.

The change in v1.8.1 helped considerably. I no longer "hang" as much when connecting to the app. Its not perfect, but definitely a major improvement.

I agree with the sentiment that it's fixed well enough for now; I no longer consider it an issue. Appreciate the follow up!