Oppzippy/OpenSCQ30 #103: Q45 - Stuck on "Loading"

Offerel, 2023-10-23:
I have tried the Android app with Q45. I can select the device in the first screen, but when i click it, the app is stuck on "Loading". Would be nice, if this headset can be supported to. Many thx.

Oppzippy, 2023-10-24:
The Q45 isn't supported yet, so one of the following happens:
1. The Q45 understands the initial request for the current sound mode, anc mode, etc. and responds, but the application does not understand the response
2. The Q45 does not understand the initial request

Either way, it's not going to work as is. I started on de obfuscating Q45 part official app, but I still have a few thousand lines to go, so it'll be some time.

As for getting stuck on loading, I must have never implemented retry/timeout on android (pretty sure I did on desktop/web). I'll do that soon and release v1.8.1.

adventure-tense, 2024-12-03:
I look forward to being able to use this with my Q45 as well. A great project.

yochananmarqos, 2025-01-12:
It's called [Space Q45 (A3040011)](https://www.soundcore.com/products/space-q45-a3040011?variant=41956218110142). @Oppzippy Maybe update the issue title for better visibility?

Oppzippy, 2025-11-09:
You don't have all of the options shown in this screenshot available on the Linux gui, but you do on Android, you're saying? Just to confirm, since I don't think that should be able to happen, unless you selected the wrong model for your device or something.

[image]

closed 2025-11-19 by Oppzippy