Oppzippy/OpenSCQ30 #105: Add support for Life P3

FineFindus, 2023-10-31:
It would be nice to see support for the [Life P3](https://us.soundcore.com/products/a3939012) as well. There was an [unfinished attempt](https://github.com/binoverfl0w/SoundCoreLifeP3) to discover the API, maybe it can be used as a starting point?

Oppzippy, 2024-01-03:
Support for the Life P3 is added in v1.10.0. I don't own one, so please let me know if it's working okay.

FineFindus, 2024-01-03:
Thanks. Sadly I'm unable to test it, since the app crashes when opened. 

```
FATAL EXCEPTION: main
Process: com.oppzippy.openscq30, PID: 31672
java.lang.UnsatisfiedLinkError: Can't obtain peer field ID for class com.sun.jna.Pointer
	at com.sun.jna.Native.initIDs(Native Method)
	at com.sun.jna.Native.<clinit>(Unknown Source:606)
	at com.sun.jna.Native.h(Unknown Source:0)
	at com.sun.jna.v0.setAlignType(Unknown Source:8)
	at com.sun.jna.v0.<init>(Unknown Source:21)
	at com.oppzippy.openscq30.lib.bindings.RustCallStatus.<init>(Unknown Source:0)
	at com.oppzippy.openscq30.lib.bindings.Openscq30_androidKt.initNativeLogging(Unknown Source:4)
	at com.oppzippy.openscq30.OpenSCQ30Application.<init>(Unknown Source:15)
	at java.lang.Class.newInstance(Native Method)
	at android.app.AppComponentFactory.instantiateApplication(AppComponentFactory.java:76)
	at androidx.core.app.CoreComponentFactory.instantiateApplication(Unknown Source:0)
	at android.app.Instrumentation.newApplication(Instrumentation.java:1160)
	at android.app.LoadedApk.makeApplication(LoadedApk.java:1325)
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:7173)
	at android.app.ActivityThread.access$2200(ActivityThread.java:296)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2208)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:213)
	at android.app.ActivityThread.main(ActivityThread.java:8178)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
```

Oppzippy, 2024-01-03:
My bad, the optimized release build was broken due to a configuration oversight. Fixed in v1.10.1.

FineFindus, 2024-01-04:
We all love proguard optimizations, don't we :)
I'm now able to open the app and select the headphones, but they just load for a while, before closing the page. 

```
1704359231.598 11503 12828 12870 D mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
1704359231.654 11503 12828 12828 I HwViewRootImpl: removeInvalidNode all the node in jank list is out of time
1704359236.821 11503 12828 12828 W DeviceConnectionManager: error connecting to device:
1704359236.821 11503 12828 12828 W DeviceConnectionManager: java.util.concurrent.TimeoutException: Timeout waiting for GATT services
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at n3.q.a(Unknown Source:569)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at n3.n.invokeSuspend(Unknown Source:12)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at v4.a.resumeWith(Unknown Source:8)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at q5.s.s(Unknown Source:6)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.a.resumeWith(Unknown Source:22)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at v4.a.resumeWith(Unknown Source:31)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at b5.i.s(Unknown Source:50)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.h.n(Unknown Source:75)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.h.z(Unknown Source:66)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.j.k(Unknown Source:66)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.f1.V(Unknown Source:25)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.f1.u(Unknown Source:179)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.s1.run(Unknown Source:31)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at android.os.Handler.handleCallback(Handler.java:888)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at android.os.Handler.dispatchMessage(Handler.java:100)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at android.os.Looper.loop(Looper.java:213)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at android.app.ActivityThread.main(ActivityThread.java:8178)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at java.lang.reflect.Method.invoke(Native Method)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: Caused by: l5.r1: Timed out waiting for 4000 ms
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	at l5.s1.run(Unknown Source:28)
1704359236.821 11503 12828 12828 W DeviceConnectionManager: 	... 7 more
1704359236.947 11503 12828 12828 I HwViewRootImpl: removeInvalidNode all the node in jank list is out of time
1704359236.962 11503 12828 12828 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1704359236.966 11503 12828 12828 I chatty  : uid=11503(com.oppzippy.openscq30) identical 6 lines
1704359236.967 11503 12828 12828 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1704359238.741 11503 12828 12870 W libEGL  : EGLNativeWindowType 0x71de2ef010 disconnect failed
```

Oppzippy, 2024-01-04:
To make sure the bluetooth le gatt services and characteristics match what I'm expecting, please do the following depending on your OS:

Linux:
Connect to the Life P3 and post the output of this command:
```bash
bluetoothctl gatt.list-attributes | grep -E "/org/bluez/.*/dev_(AC_12_2F|E8_EE_CC)_" | xargs -I {} bluetoothctl gatt.attribute-info "{}"
```

Windows/Mac:
In a Chromium based browser, go to chrome://bluetooth-internals/#devices, scan for devices, and inspect your Life P3. Expand all of the service and characteristic dropdowns, and then print the page as a pdf or something.

Android:
Same method as windows/mac, but it doesn't seem to let you print chrome:// pages, so copying and pasting everything on the page or taking multiple screenshots would be fine.

FineFindus, 2024-01-04:
> Linux:
Connect to the Life P3 and post the output of this command:

I tried it, the output is empty?

This is the result from trying it on Android.
[image]

Oppzippy, 2024-01-10:
Yeah, the services I was expecting to be there do exist. I found quite a few bugs in the android app's bluetooth connection code. They're fixed as of 1.10.2, so it's possible it'll work now. If not, more logging was added to hopefully pinpoint where things are going wrong.

FineFindus, 2024-01-14:
Sadly it's still not working, though interestingly, the automation app I use now recognizes it as a new Bluetooth connection. 

Thanks for your effort :) 

```
1705267550.722 11503 24579 24605 D mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
1705267550.822 11503 24579 24579 I HwViewRootImpl: removeInvalidNode all the node in jank list is out of time
1705267552.086 11503 24579 24579 D SoundcoreDeviceConnectorImpl: Failed to start discovering services, so we must not be connected yet. Discovery should start upon connection.
1705267552.690 11503 24579 24599 D openscq30-lib: openscq30_android::soundcore_device_utils: is_soundcore_service_uuid
1705267552.692 11503 24579 24599 D openscq30-lib: openscq30_android::soundcore_device_utils: is_soundcore_service_uuid
1705267552.693 11503 24579 24599 D openscq30-lib: openscq30_android::soundcore_device_utils: is_soundcore_service_uuid
1705267552.693 11503 24579 24599 D openscq30-lib: openscq30_android::soundcore_device_utils: write_characteristic_uuid
1705267552.698 11503 24579 24599 D openscq30-lib: openscq30_android::soundcore_device_utils: read_characteristic_uuid
1705267552.708 11503 24579 24579 D openscq30-lib: openscq30_android::connection: new
1705267552.708 11503 24579 24579 D openscq30-lib: openscq30_android::device: new_soundcore_device
1705267552.945 11503 24579 24599 I SoundcoreDeviceCallbaks: mtu changed to 500, status 0
1705267553.098 11503 24579 24579 D openscq30-lib: openscq30_android::connection: add_inbound_packet
1705267553.098 11503 24579 24579 I openscq30-lib: openscq30_android::connection: got packet [9, 255, 0, 0, 1, 1, 1, 142, 0, 1, 1, 4, 4, 0, 0, 48, 50, 46, 54, 49, 48, 50, 46, 54, 49, 51, 57, 51, 57, 50, 65, 55, 70, 67, 67, 50, 70, 49, 50, 65, 67, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255, 102]
1705267553.099 11503 24579 26196 W openscq30-lib: openscq30_lib::soundcore_device::device::soundcore_device: failed to parse packet: Error(VerboseError { errors: [([255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255], Nom(MapOpt)), ([255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255], Context("custom button model")), ([1, 1, 4, 4, 0, 0, 48, 50, 46, 54, 49, 48, 50, 46, 54, 49, 51, 57, 51, 57, 50, 65, 55, 70, 67, 67, 50, 70, 49, 50, 65, 67, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255], Context("a3945 state update packet")), ([1, 1, 4, 4, 0, 0, 48, 50, 46, 54, 49, 48, 50, 46, 54, 49, 51, 57, 51, 57, 50, 65, 55, 70, 67, 67, 50, 70, 49, 50, 65, 67, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255], Nom(Alt))] })
1705267553.714 11503 24579 26196 W openscq30-lib: openscq30_lib::soundcore_device::device::soundcore_device: fetch_initial_state: didn't receive response after 1 second on try #0
1705267553.774 11503 24579 24579 D openscq30-lib: openscq30_android::connection: add_inbound_packet
1705267553.774 11503 24579 24579 I openscq30-lib: openscq30_android::connection: got packet [9, 255, 0, 0, 1, 1, 1, 142, 0, 1, 1, 4, 4, 0, 0, 48, 50, 46, 54, 49, 48, 50, 46, 54, 49, 51, 57, 51, 57, 50, 65, 55, 70, 67, 67, 50, 70, 49, 50, 65, 67, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 1, 99, 1, 82, 1, 102, 1, 84, 1, 1, 1, 0, 7, 0, 0, 0, 10, 255, 255, 0, 255, 0, 0, 0, 51, 255, 255, 255, 255, 102]
[... cut ...]
1705267555.746 11503 24579 24579 W DeviceConnectionManager: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
1705267555.746 11503 24579 24579 W DeviceConnectionManager: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
1705267555.772 11503 24579 24579 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705267555.778 11503 24579 24579 I chatty  : uid=11503(com.oppzippy.openscq30) identical 6 lines
1705267555.778 11503 24579 24579 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
```

Oppzippy, 2024-01-14:
Getting somewhere though. Now, from these logs, I have an actual packet from the device I can use to write tests with rather than trying to predict what it will look like.

FineFindus, 2024-01-17:
```
1705531316.064 11503 31528 31528 I zippy.openscq3: Reinit property: dalvik.vm.checkjni= false
1705531316.068 11503 31528 31528 E zippy.openscq3: Not starting debugger since process cannot load the jdwp agent.
1705531316.076 11503 31528 31528 E libc    : Access denied finding property "runtime.mmitest.isrunning"
1705531316.080 11503 31528 31528 D ActivityThread: Attach thread to application
1705531316.147 11503 31528 31547 I HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=com.oppzippy.openscq30#11503#256
1705531316.147 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::Init
1705531316.147 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::StartWatch
1705531316.147 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::WatchPackage: /data/hotpatch/fwkhotpatch/
1705531316.148 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::CheckAndWatchPatch: /data/hotpatch/fwkhotpatch/com.oppzippy.openscq30
1705531316.148 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::CheckAndWatchPatch: /data/hotpatch/fwkhotpatch/all
1705531316.148 11503 31528 31528 I zippy.openscq3: QarthPatchMonintor::Run
1705531316.149 11503 31528 31550 I zippy.openscq3: QarthPatchMonintor::Reading
1705531316.149 11503 31528 31550 I zippy.openscq3: QarthPatchMonintor::CheckNotifyEvent
1705531316.149 11503 31528 31550 I zippy.openscq3: QarthPatchMonintor::CheckNotifyEvent before read
1705531316.149 11503 31528 31547 I HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=com.oppzippy.openscq30#11503#0
1705531316.179 11503 31528 31547 I AwareBitmapCacher: init processName:com.oppzippy.openscq30 pid=31528 uid=11503
1705531316.182 11503 31528 31553 E AwareLog: AtomicFileUtils: readFileLines file not exist: android.util.AtomicFile@4824da2
1705531316.183 11503 31528 31553 E AwareLog: AtomicFileUtils: readFileLines file not exist: android.util.AtomicFile@c6e9a33
1705531316.213 11503 31528 31528 V ActivityThread: callActivityOnCreate
1705531316.215 11503 31528 31528 E zippy.openscq3: Invalid ID 0x00000000.
1705531316.219 11503 31528 31528 V HwWidgetFactory: : successes to get AllImpl object and return....
1705531316.224 11503 31528 31528 I OverScrollerOptimization: start init SmartSlideOverScroller and get the overscroller config
1705531316.224 11503 31528 31528 I OverScrollerOptimization: get the overscroller config
1705531316.236 11503 31528 31528 W WindowDecorActionBar: should not do the transition or the transition anim is null or it is running or the mContainer view is null or mContainer view hasn't been drawn to screen
1705531316.241 11503 31528 31528 D HwGalleryCacheManagerImpl: mIsEffect:false
1705531316.259 11503 31528 31528 D ActivityThread: add activity client record, r= ActivityRecord{afaccbd token=android.os.BinderProxy@55ba1ee {com.oppzippy.openscq30/com.oppzippy.openscq30.MainActivity}} token= android.os.BinderProxy@55ba1ee
1705531316.281 11503 31528 31557 W HiTouch_HiTouchSensor: depended package hiTouch does n't exist!
1705531316.281 11503 31528 31557 I HiTouch_HiTouchSensor: HiTouch restricted: system app HiTouch don't exist.
1705531316.281 11503 31528 31557 D HiTouch_PressGestureDetector: onAttached, package=com.oppzippy.openscq30, windowType=1, mHiTouchRestricted=true
1705531316.332 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.338 11503 31528 31528 I chatty  : uid=11503(com.oppzippy.openscq30) identical 6 lines
1705531316.339 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.375 11503 31528 31554 I iGraphics: [0020080c] pn: com.oppzippy.openscq30, p: 31528
1705531316.375 11503 31528 31554 I iGraphics: [0030080c] no spt app: com.oppzippy.openscq30
1705531316.393 11503 31528 31554 D mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
1705531316.437 11503 31528 31554 W Gralloc3: mapper 3.x is not supported
1705531316.458 11503 31528 31554 E         : APS:IFLoad:importExternalFunctions, search function createNewHwApsUtils failed, dlsym err:undefined symbol: createNewHwApsUtils
1705531316.458 11503 31528 31554 D         : APS:importExternalFunctions OK
1705531316.461 11503 31528 31528 I HwViewRootImpl: removeInvalidNode jank list is null
1705531316.470 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.470 11503 31528 31528 I chatty  : uid=11503(com.oppzippy.openscq30) identical 1 line
1705531316.471 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.471 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.472 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.472 11503 31528 31528 I chatty  : uid=11503(com.oppzippy.openscq30) identical 2 lines
1705531316.472 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531316.766 11503 31528 31571 D HwFrameworkSecurityPartsFactory: HwFrameworkSecurityPartsFactory in.
1705531316.766 11503 31528 31571 I HwFrameworkSecurityPartsFactory: add HwFrameworkSecurityPartsFactory to memory.
[... cut ...]
1705531321.825 11503 31528 31528 I HwViewRootImpl: removeInvalidNode all the node in jank list is out of time
1705531321.828 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531321.830 11503 31528 31528 I chatty  : uid=11503(com.oppzippy.openscq30) identical 6 lines
1705531321.830 11503 31528 31528 D openscq30-lib: openscq30_android::soundcore_device_utils: is_mac_address_soundcore_device
1705531325.243 11503 31528 31554 W libEGL  : EGLNativeWindowType 0x71d1c20a50 disconnect failed
```

FineFindus, 2024-01-27:
Works in [v1.10.6](https://github.com/Oppzippy/OpenSCQ30/releases/tag/v1.10.6), so far everything seems to work as intended, if I find something that doesn't, I'll open a new issue. Thanks for your work on this :tada:

closed 2024-01-27 by FineFindus