Oppzippy/OpenSCQ30 #106: rust panic

mastermax1988, 2023-11-19:
Running arch linux with  rustc 1.74.0 stable. The gui version is not showing devices and the cli version panics:
`openscq30_cli -V          
openscq30_cli 1.8.0

openscq30_cli list-devices

thread 'main' panicked at /home/max/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.33.0/src/runtime/blocking/shutdown.rs:51:21:
Cannot drop a runtime in a context where blocking is not allowed. This happens when a runtime is dropped from within an asynchronous context.
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

Oppzippy, 2023-11-23:
I fixed the CLI panic on the master branch a few days ago. I'll do a release some time after I get back from vacation in a few days.

Oppzippy, 2023-11-27:
The CLI panic is fixed in v1.9.0.

Regarding the issue with the GUI, do you get any toast notifications in the application such as "Other(DbusError(D-Bus error: Failed (org.bluez.Error.Failed)))"? If not, try running the gui in a terminal so stdout can be seen, and check if any sort of errors are logged.

Another possibility is that there is another mac address prefix besides the two that I am aware of (AC:12:2F and E8:EE:CC). Does your device's mac address start with something other than one of those two?

Also, which device is it that you're trying to use? While only the ones [listed here](https://github.com/Oppzippy/OpenSCQ30#supported-devices) will work, all anker/soundcore devices are intended to show in the GUI's dropdown and cli's `list-devices`.

Oppzippy, 2024-01-03:
The panic is fixed and no response regarding the GUI issue, so I'm going to close this.

closed 2024-01-03 by Oppzippy