Oppzippy/OpenSCQ30 #107: Device BLE Service Not Found

bsasealteam6, 2023-11-30:
Trying to connect to Liberty 4 NC from Arch Linux with the AUR gui package, and it tells me that "Device BLE Service Not Found"

Oppzippy, 2023-11-30:
The Liberty 4 NC isn't supported yet. The list of supported devices is here: https://github.com/Oppzippy/OpenSCQ30#supported-devices

bsasealteam6, 2023-11-30:
Missed that, sorry

Kreuger, 2024-11-29:
@Oppzippy Are you planning to add support?

Oppzippy, 2024-11-29:
At some point, yes, but it won't be soon. Basically, some devices are easier to reverse engineer without actually owning them than others, so I intend to prioritize the easier ones, and this isn't one of the easier ones.

Jenna-she-her, 2025-01-11:
> At some point, yes, but it won't be soon. Basically, some devices are easier to reverse engineer without actually owning them than others, so I intend to prioritize the easier ones, and this isn't one of the easier ones.

Do you have an instruction for reverse-engineering the device.
I own it and can build for Linux and Android to test it.

Oppzippy, 2025-01-12:
Not yet, but I plan to write documentation and make some tooling for that soon.

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

Ireozar, 2025-10-31:
@Oppzippy Here it is:
```
Some(Packet { command: Command([1, 1]), body: [1, 0, 5, 5, 0, 0, 48, 53, 46, 56, 56, 48, 53, 46, 56, 56, 51, 57, 52, 55, 66, 54, 54, 50, 70, 68, 67, 67, 69, 69, 69, 56, 48, 48, 46, 48, 48, 254, 254, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 1, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 104, 100, 34, 64, 0, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 0, 0, 18, 17, 102, 17, 102, 17, 50, 17, 51, 17, 68, 17, 68, 0, 33, 0, 32, 53, 2, 80, 1, 0, 1, 1, 0, 0, 0, 0, 0, 255, 1, 2, 49, 1, 0, 0, 1, 0, 0, 1, 0, 1, 80, 1, 0, 0, 1, 1, 1, 1, 255] })
```
If you need any more just let me know, I should be mostly available today.

Ireozar, 2025-10-31:
The above was with only one bud connected, here it is with both (and noise cancellation active):
```
Some(Packet { command: Command([1, 1]), body: [1, 1, 5, 5, 0, 0, 48, 53, 46, 56, 56, 48, 53, 46, 56, 56, 51, 57, 52, 55, 66, 54, 54, 50, 70, 68, 67, 67, 69, 69, 69, 56, 48, 48, 46, 48, 48, 254, 254, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 1, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 104, 100, 34, 64, 0, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 112, 117, 140, 148, 150, 142, 134, 131, 60, 60, 0, 0, 18, 17, 102, 17, 102, 17, 50, 17, 51, 17, 68, 17, 68, 0, 33, 0, 32, 53, 0, 80, 1, 0, 1, 1, 0, 0, 0, 0, 0, 255, 1, 2, 49, 1, 0, 0, 1, 0, 0, 1, 0, 1, 80, 1, 0, 0, 1, 1, 1, 1, 255] })
```

Kreuger, 2025-10-31:
What did you guys choose? Because I dont see the 4NC in the list.

Kreuger, 2025-10-31:
Thanks. Heres mine.

>Some(Packet { command: Command([1, 1]), body: [1, 1, 5, 5, 0, 0, 48, 52, 46, 56, 56, 48, 52, 46, 56, 56, 51, 57, 52, 55, 48, 65, 50, 65, 67, 48, 67, 67, 69, 69, 69, 56, 48, 48, 46, 48, 48, 17, 0, 150, 140, 110, 110, 130, 150, 160, 170, 120, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 2, 0, 158, 148, 120, 121, 115, 115, 114, 116, 60, 60, 158, 148, 120, 121, 115, 115, 114, 116, 60, 60, 103, 38, 157, 12, 0, 158, 148, 120, 121, 115, 115, 114, 116, 60, 60, 158, 148, 120, 121, 115, 115, 114, 116, 60, 60, 0, 0, 18, 17, 97, 17, 96, 17, 50, 17, 51, 17, 66, 17, 67, 0, 255, 0, 255, 51, 0, 16, 0, 1, 0, 0, 0, 0, 0, 0, 0, 255, 1, 3, 49, 0, 0, 0, 1, 0, 1, 0, 0, 0, 90, 0, 0, 1, 1, 2, 1, 1, 255] })

Kreuger, 2025-10-31:
Out of curiousity, is there anything you could glean from Gadgetbridge's code? They support this and other Soundcore devices.

Oppzippy, 2025-11-01:
I could if necessary, but I'm not sure it would really save any time. I have the workflow down with my current method. 4835a5b667f1b4e0006405e00fd6edf196f118af took just over an hour to figure out, for example, and that covers all features supported by the device, whereas gadgetbridge only parses a small subset of that.

ozer-4, 2025-11-07:
```
Some(Packet { command: Command([1, 1]), body: [0, 1, 5, 5, 0, 0, 48, 56, 46, 53, 57, 48, 56, 46, 53, 57, 51, 57, 52, 55, 101, 56, 50, 54, 99, 102, 97, 52, 101, 57, 57, 48, 48, 46, 49, 46, 54, 254, 254, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 1, 121, 117, 120, 126, 139, 148, 147, 150, 60, 60, 121, 117, 120, 126, 139, 148, 147, 150, 60, 60, 105, 2, 3, 188, 1, 121, 117, 120, 126, 139, 148, 147, 150, 60, 60, 121, 117, 120, 126, 139, 148, 147, 150, 60, 60, 0, 0, 18, 17, 97, 17, 96, 17, 53, 17, 54, 17, 66, 17, 67, 0, 244, 0, 244, 55, 2, 80, 1, 1, 1, 1, 0, 0, 0, 0, 0, 255, 255, 4, 110, 0, 0, 0, 1, 0, 0, 1, 1, 0, 90, 0, 1, 1, 1, 2, 1, 1, 255] })
```

Ireozar, 2025-11-09:
@Oppzippy 
Results after skimming through the options:
- Sound Modes seems to be working perfectly, have not found any issues there
- Equalizer tab is completely empty when only one bud is connected, maybe it could be worth it to put some kind of message there
- Equalizer seems to be showing the default unconfigured options when using HearID, also 3D Surround Sound is missing (though no idea how many people even use that one)
- Button Configuration seems to be working perfectly
- Limit High Volume works
- Miscellaneous only has very few options, I'd assume that there could be a bit more, especially (though that might be wrong) configuration of:
	- Auto Play/Pause
	- Gaming Mode
	- Sound Leak Compensation
	- more Prompt Tones (currently there's only Touch Tone, missing Wearing and Low Battery Prompt Tones)
	- Sound Mode (though the time it supposedly takes to restart seems scary)
	- Dual Connections seems unlikely to me
	- Find Device would probably be a different communication protocol? (but maybe not, just feels like it could be)

All in all quite usable though, thanks so much for your work!

Jenna-66, 2025-11-09:
I can't test at the moment. I'm not at home (most likely for at least two more weeks) and forgot the earphones.
🤦‍♀️😢

Oppzippy, 2025-11-19:
All missing options except for dual connection and find device are added in v2.0.0-beta6.

I would need wireshark logs for enabling/disabling dual connections, since I'm not sure how that works (and I ought to write instructions for how to capture logs at some point). If someone wants to do this, please open another issue for it, since this applies to all devices supporting dual connections and isn't Liberty 4 NC specific.

I could be mistaken, but Find Device seems to just play audio normally as you would any other audio. Turning down and up the volume on my phone affects the Find Device volume, for example. If that is the case, I don't really care to add support for it myself, since it means using platform specific audio apis, and the functionality can be replicated without OpenSCQ30 just by playing a similarly loud audio file. If someone in the future submitted a PR for it, I wouldn't be opposed to it though.

Oppzippy, 2025-12-05:
> Id also like the option to show as a percentage

It shows as a percentage as of [v2.0.0-beta7](https://github.com/Oppzippy/OpenSCQ30/releases/tag/v2.0.0-beta7).

SammysHP, 2026-09-25:
Here are a few options that are missing for the Liberty 4 NC:

- LDAC toggle
- Dual connections
- Button config: available modes (normal/transparency/NC)