Oppzippy/OpenSCQ30 #108: P20i

kodoor, 2023-12-03:
Any chance of getting support for the P20i??

Oppzippy, 2023-12-05:
At some point, yes, but it won't be soon.

kodoor, 2024-01-10:
Is their any information I could get for you to help with support?

I picked up some to try as 10 hours usage seems good for the price but the spyware in the app is to much...

Ryoma123456789, 2024-08-06:
Is there any news on this support?

Oppzippy, 2024-08-06:
No, I haven't been putting as much time into this project as I used to, so things are going slowly. #132 is what I'm working on now. After that, support for more devices is next.

quelbs, 2026-01-01:
State update packet for p20i:
```
Some(Packet { command: Command([1, 1]), body: [1, 0, 255, 4, 0, 0, 0, 0, 0, 0, 0, 49, 52, 46, 52, 51, 51, 57, 52, 57, 66, 57, 65, 69, 67, 55, 50, 67, 57, 67, 49, 56, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 14, 1, 255, 1, 255, 1, 245, 1, 102, 1, 242, 1, 51, 255, 255, 255, 97, 0, 1, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255] })
```