Oppzippy/OpenSCQ30 #111: Auto start on device connect / Auto stop on disconnect?

Mrnofish, 2023-12-09:
Most manufacturer apps detect whenever their device connects/disconnect, thus display/hide the relevant notification, saving the user the hassle to run/terminate the app manually.

This would also more or less require the app to acquire the ability to connect automatically, instead of relying on the user selecting the device manually on every start-up.

Not sure how much work this might entail, but I see there is already a DeviceService so perhaps not an outlandish idea, and it would address the major 'workflow' hiccup in the app.

Oppzippy, 2023-12-09:
Yeah, this sounds reasonable to have as an option. It should be fairly straight forward for me to add.

SunDevil311, 2024-03-17:
You may have added this functionality already, but if not, this can already be set up to some extent on Samsung phones via Modes & Routines.

I've attached a screenshot of the routine I have set up to launch OpenSCQ30 whenever my Life Q35 headphones connect via Bluetooth.

Thought it might be useful for some.

[image]

Oppzippy, 2024-04-21:
It was not nearly as straight forward as I thought, but this is implemented now. On the device selection screen, there is a settings button with an option to enable auto connect.

There are two known situations where it won't work:
1. On boot, if the device connects to the phone before openscq30's background service launches, it will not auto connect.
2. If android kills the background service due to high ram usage or whatever, it will not receive the events to launch the foreground service and connect. The background service will start again when system resources are available, however.

Other than that, it should work fine.

closed 2024-04-21 by Oppzippy

SunDevil311, 2024-04-21:
@Oppzippy That is exciting news! Sounds like a fairly major engineering achievement. Aside from the firmware update option (which I doubt gets much use after first run), I'm pretty sure you're at feature parity and then some with Soundcore.

Thanks for your hard work on this!