Oppzippy/OpenSCQ30 #122: Life P3 error

ayathedoggo, 2024-01-20:
Hullo. I'm not technically inclined, first time posting on Github, be nice pls

I got a block of code in command prompt that the specified file tried to run was not found; after a few seconds, it opened. Me installing it somewhere other than the default folder, maybe. In any case, upon finding my device, connected via Bluetooth, it returned "Device BLE characteristic not found". Is there any way this can be fixed? I'd love to learn too

```
 ←[2m2024-01-20T18:13:44.796044Z←[0m ←[33m WARN←[0m  ←[33minitial load of settings file failed: Failed to load config

Caused by:
    0: get config file
    1: open file C:\Users\User\AppData\Roaming\OpenSCQ30\config.toml for Read
    2: The system cannot find the file specified. (os error 2)←[0m
    ←[2;3mat←[0m gui\src\main.rs:195

  ←[2m2024-01-20T18:13:53.527482Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    characteristic `00008888-0000-1000-8000-00805f9b34fb` not found: None←[0m
    ←[2;3mat←[0m gui\src\main.rs:273

  ←[2m2024-01-20T18:14:26.366412Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    characteristic `00008888-0000-1000-8000-00805f9b34fb` not found: None←[0m
    ←[2;3mat←[0m gui\src\main.rs:273
```

Oppzippy, 2024-01-20:
Thank you for reporting this.

I should probably change the log message about the config file to `info` instead of `warn`. There's not anything wrong, it just means that you haven't done anything yet within OpenSCQ30 that would cause the creation of a config file.

The "Device BLE characteristic not found" is surprising, I'll look into it.

Oppzippy, 2024-01-27:
By any chance, were your earbuds connected to more than one device at the same time? For example, both a phone and your computer. That's a known issue: #90 

If not, does refreshing and trying to connect again help?

Also, a few Life P3 bugs were fixed in v1.10.6 (just released a few days ago), although none related to the initial connection failing on Windows like this. So while it won't fix this specific issue, upgrading will fix some others that you would likely encounter after connecting successfully.

ayathedoggo, 2024-01-30:
Hey! Sorry for the late response, I don't use Github and am terrible about my emails.

The Life P3s (unfortunately) do not support multipoint connection (i.e. connecting to multiple devices).

I'll check

ayathedoggo, 2024-01-30:
Can confirm the program seems a bit unhappy with me. 😅

Installed the new version, launched, it found the device, went through, it went to a "Loading..." screen, then did not respond. Tried waiting, crashed again. Clicked setup again, installed other files, booted up...

It found my (connected) device, returns a 404
[image]
[image]

ayathedoggo, 2024-01-30:
```
  ←[2m2024-01-30T14:39:20.851968Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    0: device not found: Error { code: HRESULT(0x00000000), message: "The operation completed successfully." }
    1: The operation completed successfully. (0x00000000)←[0m
    ←[2;3mat←[0m gui\src\main.rs:273

  ←[2m2024-01-30T14:39:26.329315Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    0: device not found: Error { code: HRESULT(0x00000000), message: "The operation completed successfully." }
    1: The operation completed successfully. (0x00000000)←[0m
    ←[2;3mat←[0m gui\src\main.rs:273

  ←[2m2024-01-30T14:39:26.730252Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    0: device not found: Error { code: HRESULT(0x00000000), message: "The operation completed successfully." }
    1: The operation completed successfully. (0x00000000)←[0m
    ←[2;3mat←[0m gui\src\main.rs:273

  ←[2m2024-01-30T14:43:31.108945Z←[0m ←[31mERROR←[0m  ←[31mselect device

Caused by:
    0: device not found: Error { code: HRESULT(0x00000000), message: "The operation completed successfully." }
    1: The operation completed successfully. (0x00000000)←[0m
    ←[2;3mat←[0m gui\src\main.rs:273

```

Oppzippy, 2024-02-01:
Does the web version work (https://oppzippy.github.io/OpenSCQ30/, requires a Chromium based browser for Web Bluetooth)? Knowing if it does or not would help in narrowing down the problem.

ayathedoggo, 2024-02-02:
Web browser does work, yes. _(wow, this is awesome)_
Had to turn on Web Bluetooth API in brave://flags/.