Oppzippy/OpenSCQ30 #123: Life Q30 failing to parse packets

matanton666, 2024-01-22:
Hi! im pretty new here so mabe this is an easy fix i just dont understand.
I am trying to connect to my Life Q30 headset but it keeps failing when i press the connect button. 
this is the log i get:
 ```
 2024-01-22T18:49:05.935418Z  INFO  Services already resolved.
    at /run/build/openscq30_gui/cargo/vendor/bluez-async-0.7.2/src/lib.rs:696

  2024-01-22T18:49:06.175122Z  WARN  failed to parse packet: Error(VerboseError { errors: [([69, 56], Nom(MapOpt)), ([67, 67, 69, 69, 69, 56], Context("custom button model")), ([4, 0, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 255, 255, 255, 48, 51, 46, 53, 48, 51, 48, 50, 56, 65, 50, 51, 67, 52, 52, 67, 67, 69, 69, 69, 56], Context("a3945 state update packet")), ([4, 0, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 255, 255, 255, 48, 51, 46, 53, 48, 51, 48, 50, 56, 65, 50, 51, 67, 52, 52, 67, 67, 69, 69, 69, 56], Nom(Alt))] })
    at lib/src/soundcore_device/device/soundcore_device.rs:144

  2024-01-22T18:49:07.117029Z  WARN  fetch_initial_state: didn't receive response after 1 second on try #0
    at lib/src/soundcore_device/device/soundcore_device.rs:157

  2024-01-22T18:49:07.175102Z  WARN  failed to parse packet: Error(VerboseError { errors: [([69, 56], Nom(MapOpt)), ([67, 67, 69, 69, 69, 56], Context("custom button model")), ([4, 0, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 255, 255, 255, 48, 51, 46, 53, 48, 51, 48, 50, 56, 65, 50, 51, 67, 52, 52, 67, 67, 69, 69, 69, 56], Context("a3945 state update packet")), ([4, 0, 0, 0, 120, 120, 120, 120, 120, 120, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 255, 255, 255, 48, 51, 46, 53, 48, 51, 48, 50, 56, 65, 50, 51, 67, 52, 52, 67, 67, 69, 69, 69, 56], Nom(Alt))] })
    at lib/src/soundcore_device/device/soundcore_device.rs:144
```

it logs the same thing 2 more times (3 trys of connecting) and then gives:
```
 2024-01-22T18:49:09.124713Z ERROR  select device

Caused by:
    device didn't respond to request
    at gui/src/main.rs:273

```

i am using the newest flatpack version of the app on debian.
any help would be appreciated! thanks.

Oppzippy, 2024-01-22:
Interesting, looks like your Q30 is running firmware version 03.50, but the newest available for mine is 02.30. I guess it's probably a newer model or something with slightly different behavior. Looking into it.

Mrnofish, 2024-01-22:
FWIW I just installed the latest Windows build to see what would happen, and it's working for me.

Mine came with 3.5 from factory.

Oppzippy, 2024-01-23:
@Mrnofish Thanks, that's useful to know and possibly leads me to the answers all my questions.

The Q30 does not have a custom transparency mode (vocal mode/fully transparent) and custom noise canceling strength setting, so those values are usually valid but ignored. None of the valid values for those settings exceed 0x0A, but @matanton666's Q30 is reporting 0xFF for the noise canceling mode, custom transparency mode, and custom noise canceling settings. I assumed 0xFF was supposed to be null essentially, and the device would override whatever its actual value is and report 0xFF when the setting is not applicable. However, it's possible that the device just accepts whatever value you throw at it, and 0xFF just happened to end up getting set for whatever reason and it's unrelated to the firmware version. I'll check if this is the case later.

Oppzippy, 2024-01-24:
Should be fixed in v1.10.5. https://github.com/flathub/com.oppzippy.OpenSCQ30/pull/13 is merged, so it shouldn't be too long until it's available on flathub. Please let me know if the issue still occurs.

closed 2024-01-24 by Oppzippy