Oppzippy/OpenSCQ30 #126: Life P3 working

myanesp, 2024-02-09:
Hi there,

Thanks for your amazing work! As the README says that P3 status is unknown, I can confirm that is working at least on Android app and webversion.

The only thing is that if some time has passed since the connection, OpenSCQ30 won't connect to the device. I don't know if this a known issue; anyway, just with opening the app just after the Bluetooth connection with the phone everything works.

I'll try this weekend to test on Windows and Linux, and if it is possible, with Mac.

Thanks again

myanesp, 2024-02-29:
Sorry for the delay.

Working both on Windows and Linux. I saw that you put already working on the README! Anyway, if you're okay, I'll be this issue open until I (or anyone else) test on Mac. I'll try to do it in the next week.

Cheers

Oppzippy, 2025-09-29:
closing since macos support is dropped in 2.0.0-beta1