Oppzippy/OpenSCQ30 #131: Bug/Error: Appimage file fails to load

EldrinSMP, 2024-05-10:
ISSUE:
When launching the openscq30_gui-1.12.0-x86_64.AppImage version fo the app it fails to load. Activating the terminal, and I see that it throws the following error:

`/tmp/.mount_openscruBWUq/usr/bin/openscq30_gui: symbol lookup error: /tmp/.mount_openscruBWUq/usr/bin/openscq30_gui: undefined symbol: adw_navigation_view_get_type, version LIBADWAITA_1_0`

EXPECTED BEHAVIOR:
OpenSCQ30 program opens without issue. 

STEPS TO REPRODUCE:
1. Download openscq30_gui-1.12.0-x86_64.AppImage from release page of this repository.
2. Execute the file.

SYSTEM INFORMATION:
Irrelevant, it's an AppImage but just in case:
ArchLinux
Kernel 6.8.9-arch1-2 (64-bit)
KDE Plasma 6.0.4

COMMENTS:
As I undersetan AppImages, they should be self contained and have packaged everything needed to run the application, however when something is missing it will reach out to the host system to find what it needs. In my case, I have the libadwaita package on my system, but it's version 1:1.5.0-1. It looks like the appimage is exepcting  1.0.  I'm going to open up the file and see what I can find.

Oppzippy, 2024-05-11:
The problem is that OpenSCQ30 has required libadwaita 1.4 for quite a while, but I never updated the appimage from ubuntu 22.04, which had something older than that. I use appimage-builder for generating the appimages, but I'm having problems getting that to work with ubuntu 24.04, so I'll either have to figure that out, or switch to something else since appimage-builder seems unmaintained.

In the mean time, I'd recommend using either the flatpak or @SpacingBat3's AUR package.

EldrinSMP, 2024-05-14:
I appreciate the response. I tried the AUR package and I'm having different issues with it, not sure if it's my headphones, system, or the app so I'm hesitant to file an issue for it. I haven't had a chance to research the error yet. I'll check out the Flatpak as well to see if I get the same issues. Thanks!

closed 2025-10-31 by Oppzippy