Oppzippy/OpenSCQ30 #134: Any Chance of Supporting Liberty 3 Pro?

rebeltaz, 2024-07-10:
I didn't see it on the list, but tried anyway and it times out waiting for a reply. I assume that that is because it's not supported? I have been looking for something like this. It sucks that manufacturers neglect to support PC users.

Oppzippy, 2024-07-13:
It's on my to do list. It won't be soon, but I do intend to add support for it at some point.

NachtigallersLehrling, 2024-07-13:
In case you need inspiration with the protocol:
I recently added support for the Liberty 3 Pro to the Gadgetbridge Project (partly inspired by your codebase).
No warranty, but it works for me.
https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3753

whysofurious1, 2024-12-27:
Just to add that Gadgetbridge added support for Libery 4 Pro as well (https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/4050) because of basically identical implementation (not an expert tough), so when time comes it might be worth to add support for both :)

Tanks in advance!

Oppzippy, 2026-04-29:
Support for the Liberty 3 Pro is added in v2.7.0. Let me know if there's anything not working.