Oppzippy/OpenSCQ30 #140: soundcore space one

lovaye1, 2024-11-25:
please add soundcore space one too

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

CummingCowGirl, 2025-11-11:
I'm assuming this is for Space One Pro?

Oppzippy, 2025-12-05:
No, the space one pro is different

SpiZeak, 2025-12-18:
Looking forward to seeing this implemented!

David112x, 2026-01-04:
> > Support for the space one is added in v2.2.0. Let me know if there are any issues.
> 
> <img alt="Image" width="1024" height="768" src="https://private-user-images.githubusercontent.com/82758916/531698607-b7b328d6-0d21-4d34-9b12-12d63346bd87.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc1MDU3NzQsIm5iZiI6MTc2NzUwNTQ3NCwicGF0aCI6Ii84Mjc1ODkxNi81MzE2OTg2MDctYjdiMzI4ZDYtMGQyMS00ZDM0LTliMTItMTJkNjMzNDZiZDg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA0VDA1NDQzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNiMGJjMzA0ZThlNDlhOGM2Njk4YTgzMWY3Y2NiNTIxNjU1YWU2MTg0M2ZlNDg1MTE1Mjk4MTAyODM1OGQ0OWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.hfuPcqLnfOuMfrg_nRH2Lgu-PdkH2DDy3rUXSl4VXkc">
> Any idea about this? tried opening as Administrator and getting the same issue, terminal won't show up either so I can't exactly copy the error.
> 
> EDIT: Tried the installer version, got the terminal and got this error while trying to add a device:
> 
> ```
>   2026-01-04T05:43:30.049505Z ERROR openscq30_gui::add_device: SoundcoreA3035 fetching devices: ConnectionError { source: Other { source: Error { code: HRESULT(0x8000000E), message: "A method was called at an unexpected time." }, location: Location { file: "D:\\a\\OpenSCQ30\\OpenSCQ30\\lib\\src\\connection_backend\\windows\\rfcomm.rs", line: 35, column: 27 } }, location: Location { file: "/rustc/ded5c06cf21d2b93bffd5d884aa6e96934ee4234\\library\\core\\src\\ops\\function.rs", line: 250, column: 5 } }
>     at gui\src\add_device.rs:252
> ```
> 
> Not specific to Space One.
> 
> cc [@Oppzippy](https://github.com/Oppzippy)

Update 2: This might be an issue with my machine specifically? Using Windows 11 (24H2) on a laptop, this same issue occurs with 2.1.0.

Oppzippy, 2026-01-04:
Not a problem on your end, the openscq30 windows bluetooth backend is indeed broken. I develop on linux and usually only test the windows bluetooth backend whenever I touch it, but I must have forgotten to last time I updated windows-rs. v2.2.1 is [building now](https://github.com/Oppzippy/OpenSCQ30/actions/runs/20689758955) with the fix and then I'll do another release.