Oppzippy/OpenSCQ30 #141: Soundcore Q30 not showing on Device list on Linux

vinoff, 2024-12-05:
I am connected to my Soundcore Q30 on my desktop and they are working fine. However, I can't seem to get OpenSCQ30 working...

The device list shows as "(None)". Clicking Refresh does nothing. Nor does "Connect"..

Any suggestions?

I am running Arch-Linux, openscq30-gui 1.14.1-1. Q30's firmware is 04.33..

Oppzippy, 2024-12-06:
What is the first half of your Q30's mac address? I had been filtering the devices by mac addresses starting with AC:12:2F or E8:EE:CC since I thought all soundcore devices were within those ranges, but I recently encountered some [other](https://github.com/Oppzippy/OpenSCQ30/commit/2ba694273de041497b1706677c62e51f893bfc18) [ranges](https://github.com/Oppzippy/OpenSCQ30/commit/888e4c52bb04348892306a863d533be6b11536ad) (not included in a release yet), so it's possible yours is outside the expected range.

vinoff, 2024-12-06:
```
> bluetoothctl
[soundcore Q30]# Agent registered
[soundcore Q30]# devices
Device 98:47:44:93:C8:6D soundcore Q30
Device E8:EE:CC:76:50:1A soundcore Liberty 4 NC
```

Seems to be 98:47:44:93:C8:6D.

Oppzippy, 2024-12-14:
It should be fully working now as of v1.16.0. Feel free to re-open if there are still problems.

vinoff, 2024-12-14:
I tested it on Android (package has not yet been updated on AUR so I cannot test it on linux yet) and it doesn't work on Android.. Same problem, the device is not found..

Oppzippy, 2024-12-14:
Now that's surprising. I use the same method as the official Soundcore app on Android for device filtering, and that's checking if the device name contains "Soundcore". Is the device name something other than "Soundcore Life Q30" for you?

vinoff, 2024-12-14:
Yes, I had changed it to "q30" in the bluetooth settings.. Now, I changed it to "Soundcore Life Q30". Dunno if that should make a difference, but it didn't. 

Here is what I see when I click on Pair:

[image]

And here is a screenshot of the official soundcore app. Seems like the device name is just "Q30".

[image]

Oppzippy, 2024-12-14:
> ```
> > bluetoothctl
> [soundcore Q30]# Agent registered
> [soundcore Q30]# devices
> Device 98:47:44:93:C8:6D soundcore Q30
> Device E8:EE:CC:76:50:1A soundcore Liberty 4 NC
> ```
> 

I forgot that this comment had the device name in it. Looks like the name does have soundcore in it, just with a lowercase s. I made the filter case insensitive in v1.16.1, so that should fix it.

vinoff, 2024-12-14:
> I forgot that this comment had the device name in it. Looks like the name does have soundcore in it, just with a lowercase s. I made the filter case insensitive in v1.16.1, so that should fix it.

That was in Linux though, not on Android...? I haven't tested v1.16.0 on Linux yet, since I am using the package from AUR (arch linux) and it has not yet been updated. The screenshots I showed were android's.

vinoff, 2024-12-14:
v1.16.1 still doesn't work on android..

Oppzippy, 2024-12-14:
> That was in Linux though, not on Android

My mistake, wasn't aware `bluetoothctl devices` output aliases and not the actual device name. So I guess the "soundcore" part was added by you, and the actual name is just "Q30" as you said. It's very strange, I have no idea why they removed the "Soundcore Life" part from the device name. I'll look into what I can do, but the solution might need to just be having a separate "Pair Device (Unfiltered)" button or something.

vinoff, 2024-12-15:
> My mistake, wasn't aware `bluetoothctl devices` output aliases and not the actual device name. So I guess the "soundcore" part was added by you, and the actual name is just "Q30" as you said. It's [...]

I don't think I added anything to the device name on Linux.. Is it possible that the device name is different depending on the OS? On Android, it is "Q30" and on Linux, it is "soundcore Q30"? Because that is what I am seeing.

I will test the v1.16.1 on Linux soon and I will let you know if it works.

Oppzippy, 2024-12-15:
It shouldn't change, the device is broadcasting its name whether or not anything is connected to it. I had a look at the Bluetooth spec and `HCI_Write_Local_Name` for changing the device's name on the device itself (not an alias on your computer/phone) does exist, but I'm pretty sure that's very different from renaming a device in the operating system's UI.

Here's what happens if I rename the device to "Q30" and scan with the "Soundcore" name filter:

[image]

Even though my alias no longer has "Soundcore" in the name, the actual name the device is broadcasting does still contain "Soundcore".

On Android, the simplest way to figure out the real name seems to be to un-pair and re-pair.

On linux:
```
❯ bluetoothctl info
Device AC:12:2F:C8:6E:08 (public)
        Name: Soundcore Life Q30
        Alias: Q30 Test Alias
```

vinoff, 2024-12-15:
This is what I get on Linux:

```
> bluetoothctl info
Device 98:47:44:93:C8:6D (public)
        Name: soundcore Q30
        Alias: soundcore Q30
```
        
On android, I unpaired and paired and the name remains Q30 in the soundcore app (`Device Name: Q30`, as the screenshot I posted earlier shows). On android, in the bluetooth settings, it says "John's soundcore Q30". 

v1.16.1 doesn't work on android. Waiting to test on Linux.

vinoff, 2024-12-16:
@Oppzippy, I just tested v1.17.0 on Android:

[image]

Clicking on "Pair Device (Unfiltered)" does find a device.. However, after pairing, I see the screenshot above.. If I click on that, nothing happens.. As in, I can't do anything with it.

Oppzippy, 2024-12-17:
A few questions:
1. Are you sure you're connecting to the right device? The mac address there starts with 61 rather than 98 like before.
2. On linux, if you run `bluetoothctl` and then `scan on`, wait a bit, and then `devices`, are there two entries for Q30 with different mac addresses? `scan on` will scan for both bluetooth low energy and non low energy devices, and I saw something in a different issue indicating that it may be possible for different mac addresses to be used for each of those.

Next, please do the following so I can ensure the BLE services are as I expect. It's possible that newer devices don't support BLE, so if you get no results, that could be the case.
1. `bluetoothctl`
2. `connect YOUR_DEVICES_MAC_ADDRESS` - this is needed even if already connected, since it will cause GATT services to be discovered, and that does not occur automatically. This should output a giant list of services and such.
4. `exit`
5. Post the output of this command:
```bash
bluetoothctl gatt.list-attributes | grep -E "/org/bluez/" | xargs -I {} bluetoothctl gatt.attribute-info "{}"
```

Oppzippy, 2024-12-17:
It's seems that you're not doing anything wrong and that they removed BLE support in newer devices, since they don't actually use that in their app. I'll need to change the implementation on all platforms except web to use standard bluetooth. Web Bluetooth only supports BLE, so it won't be possible to support newer devices on web.

Oppzippy, 2024-12-17:
One last thing to check just to make sure. On Android in a chromium based browser, navigate to chrome://bluetooth-internals. Go to the devices tab, and start scan until Q30 shows up. Then, click inspect. If there are services listed there, screenshot it. If not, well, the solution is a lot of work and won't happen soon.

Oppzippy, 2024-12-17:
It will only show bluetooth low energy devices there, so if it's not showing up, it doesn't support BLE. Unfortunate.

vinoff, 2024-12-20:
I just tested on Linux with v1.17.0 and sure enough I get the following error:

"Device BLE Service Not Found".

Oppzippy, 2025-01-08:
As of v1.18.0, bluetooth classic is used rather than bluetooth low energy, so that should fix that problem. I did see something in another issue indicating that these newer Q30's could have a different packet structure, so it is possible it will now find the service but fail to parse the state update packet.

Mrnofish, 2025-01-08:
Won't switching to regular Bluetooth on BLE-supporting devices result in a reduction in battery life ?

Oppzippy, 2025-01-08:
I haven't measured or anything, but considering the official Soundcore app doesn't make use of BLE at all, I doubt it has a noticeable impact. Audio is over bluetooth classic in either case, so it's not like we're avoiding running bluetooth classic by using BLE.

vinoff, 2025-01-09:
Tested (v1.18.0) on Android:

On the screen that says "soundcore Q30 - Paired", I click on it, it shows "Loading for a second or so and then goes back to the same screen. Nothing happens.

Mrnofish, 2025-01-09:
1.18 is working for me on Q30 and Android. However, #146 is unchanged.

vinoff, 2025-01-10:
Tested 1.18.1 on Android and it is still not working. Pressing the paired headphones, it just shows "Loading" for like 2 secs and then goes back to the main screen.

Oppzippy, 2025-01-11:
v1.18.1 was a fix for a different issue, so I wasn't expecting anything to change with this. I really should have done this ages ago, but I added a button in the settings page to copy logs to clipboard in v1.19.0. Please try to connect, and then post the error logs.

vinoff, 2025-01-11:
Here it is:
```

01-11 08:33:24.822 28152 28152 D FramePredict: FramePredict init: false
--------- beginning of main
01-11 09:40:02.172 21051 21051 I zippy.openscq30: Using CollectorTypeCC GC.
01-11 09:40:02.177 21051 21051 E zippy.openscq30: Not starting debugger since process cannot load the jdwp agent.
01-11 09:40:02.181 21051 21051 D nativeloader: Load libframework-connectivity-tiramisu-jni.so using APEX ns com_android_tethering for caller /apex/com.android.tethering/javalib/framework-connectivity-t.jar: ok
01-11 09:40:02.186 21051 21051 W System.err: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
01-11 09:40:02.186 21051 21051 W System.err: 	at libcore.io.Linux.open(Native Method)
01-11 09:40:02.186 21051 21051 W System.err: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
01-11 09:40:02.186 21051 21051 W System.err: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:274)
01-11 09:40:02.186 21051 21051 W System.err: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
01-11 09:40:02.187 21051 21051 W System.err: 	at android.app.ActivityThread$AndroidOs.open(ActivityThread.java:8381)
01-11 09:40:02.187 21051 21051 W System.err: 	at android.system.Os.open(Os.java:494)
01-11 09:40:02.187 21051 21051 W System.err: 	at android.os.perfdebug.PerfDebugMonitorImpl.monitorVersionControl(PerfDebugMonitorImpl.java:206)
01-11 09:40:02.187 21051 21051 W System.err: 	at android.os.perfdebug.PerfDebugMonitorImpl.prepareMonitor(PerfDebugMonitorImpl.java:187)
01-11 09:40:02.187 21051 21051 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:8468)
01-11 09:40:02.187 21051 21051 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
01-11 09:40:02.187 21051 21051 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:561)
01-11 09:40:02.187 21051 21051 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954)
01-11 09:40:02.187 21051 21051 I MessageMonitor: Load libmiui_runtime
01-11 09:40:02.203 21051 21051 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10743; state: ENABLED
01-11 09:40:02.225 21051 21051 D nativeloader: Configuring clns-4 for other apk /data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk. target_sdk_version=34, uses_libraries=, library_path=/data/app/lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/lib/arm64:/data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/com.oppzippy.openscq30
01-11 09:40:02.234 21051 21051 D nativeloader: Load libframework-connectivity-jni.so using APEX ns com_android_tethering for caller /apex/com.android.tethering/javalib/framework-connectivity.jar: ok
01-11 09:40:02.236 21051 21051 I Perf    : Connecting to perf service.
01-11 09:40:02.242 21051 21051 V GraphicsEnvironment: ANGLE Developer option for 'com.oppzippy.openscq30' set to: 'default'
01-11 09:40:02.242 21051 21051 V GraphicsEnvironment: ANGLE GameManagerService for com.oppzippy.openscq30: false
01-11 09:40:02.242 21051 21051 V GraphicsEnvironment: App is not on the allowlist for updatable production driver.
01-11 09:40:02.245 21051 21051 I ForceDarkHelperStubImpl: initialize for com.oppzippy.openscq30 , ForceDarkAppConfig: null
01-11 09:40:02.246 21051 21051 D nativeloader: Load libforcedarkimpl.so using system ns (caller=/system_ext/framework/miui-framework.jar): ok
01-11 09:40:02.246 21051 21051 D zippy.openscq30: JNI_OnLoad success
01-11 09:40:02.246 21051 21051 I MiuiForceDarkConfig: setConfig density:2.750000, mainRule:0, secondaryRule:0, tertiaryRule:0
01-11 09:40:02.247 21051 21051 D NetworkSecurityConfig: No Network Security Config specified, using platform default
01-11 09:40:02.247 21051 21051 D NetworkSecurityConfig: No Network Security Config specified, using platform default
01-11 09:40:02.257 21051 21051 D nativeloader: Load /data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk!/lib/arm64-v8a/libopenscq30_android.so using ns clns-4 from class loader (caller=/data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk): ok
01-11 09:40:02.260 21051 21051 D nativeloader: Load /data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk!/lib/arm64-v8a/libjnidispatch.so using ns clns-4 from class loader (caller=/data/app/~lrSn_ly2riejOuOCit6qfA==/com.oppzippy.openscq30-bn7-MO-mDvbIcvm2rM5-ig==/base.apk): ok
01-11 09:40:02.303 21051 21051 W libc    : Access denied finding property "ro.vendor.df.effect.conflict"
01-11 09:40:02.300 21051 21051 W zippy.openscq30: type=1400 audit(0.0:3920034): avc: denied { read } for name="u:object_r:vendor_displayfeature_prop:s0" dev="tmpfs" ino=2431 scontext=u:r:untrusted_app:s0:c231,c258,c512,c768 tcontext=u:object_r:vendor_displayfeature_prop:s0 tclass=file permissive=0 app=com.oppzippy.openscq30
01-11 09:40:02.307 21051 21051 D IS_CTS_MODE: true
01-11 09:40:02.307 21051 21051 D MULTI_WINDOW_ENABLED: false
01-11 09:40:02.309 21051 21051 D DecorView[]: getWindowModeFromSystem  windowmode is 1
01-11 09:40:02.312 21051 21051 D ScrollerOptimizationManager: registerConfigChangedListener
[... cut ...]
01-11 09:40:11.669 21051 21056 W zippy.openscq30: Missing inline cache for java.lang.Object T0.n.j(java.lang.Object)
01-11 09:40:13.794 21051 21051 I SettingsViewModel: exporting logs
01-11 09:40:13.798 21051 21056 W zippy.openscq30: Missing inline cache for void android.view.InputEventLogStubImpl.logOnInputEvent(android.view.InputEvent, java.lang.String)
01-11 09:40:13.799 21051 21057 I zippy.openscq30: Background concurrent copying GC freed 16MB AllocSpace bytes, 8(152KB) LOS objects, 49% free, 4740KB/9480KB, paused 164us,42us total 110.452ms
```

Oppzippy, 2025-01-11:
Thanks, fixed in v1.19.1. It seems the newer Q30's have some extra options (touch control, dual connections, auto power off, prompt tones) that I was unaware of, so that was the cause of the problem.

vinoff, 2025-01-12:
Tested on Android: It works!

Thank you! Finally a way to create an equalizer without having to guess the numbers. Thank you @Oppzippy!

closed 2025-01-12 by vinoff