Oppzippy/OpenSCQ30 #143: Soundcore Life A2 NC

platypusgit, 2024-12-15:
Version: 1.16.1

I've got the Life A2 NC and unfortunately I have to report, that the Android-App does not recognize them at the moment. 

- Paired with my phone and working
- Open up the App and click on pair device
- Pop-up opens and shows a loading animation, but no device shows up, after a while the pop-up closes automatically

I hope support can be added in the future. It's great that there's an open-source App for headphones without tracking, keep up the good work!

Oppzippy, 2024-12-15:
Probably the same problem as #141. I have it filter for devices with names containing "Soundcore", but it seems they stopped including that in the name. I have a workaround implemented, which is to have a "Pair Device (Unfiltered)" button for when the filtering doesn't work, but it hasn't been released yet. I'll do another release later today.

Oppzippy, 2024-12-17:
Please try unfiltered pairing in v1.17.0.

platypusgit, 2024-12-18:
Version: 1.17.0

- When I click on _Pair Device (Unfiltered)_ all Wifi-Networks(!) in my vicinity are shown, no bluetooth-devices.
- The headphones are correctly named "Soundcore Life A2 NC" in the Android settings. The address starts with AC:12:2F..., as expected.
- I am not able to find the device under chrome://bluetooth-internals/#devices as suggested in the other thread.
- Firmware-Version per official App: 02.20.

flue17, 2024-12-23:
I've also got Life A2 NC and tried to connect with them in the windows gui.

The headphones are connected via bluetooth.
When I click on connect the following error message appears:
[image]

`  2024-12-23T12:09:27.884230Z ERROR  select device

Caused by:
    0: device not found: Error { code: HRESULT(0x00000000), message: "Der Vorgang wurde erfolgreich beendet." }
    1: Der Vorgang wurde erfolgreich beendet. (0x00000000)
    at gui\src/main.rs:288`

Then I used the cli and got this result:

`PS C:\Users\Fabian\Downloads> .\openscq30_cli-windows-x86_64.exe -V
openscq30_cli 1.17.0
PS C:\Users\Fabian\Downloads> .\openscq30_cli-windows-x86_64.exe -l debug list-devices
AC:12:2F:35:BD:AB
PS C:\Users\Fabian\Downloads> .\openscq30_cli-windows-x86_64.exe -l debug -m AC:12:2F:35:BD:AB get noise-canceling-mode
  2024-12-23T12:19:48.726693Z DEBUG  AC:12:2F:35:BD:AB is not cached
    at lib\src\soundcore_device\device\soundcore_device_registry.rs:87

Error: "Error fetching device: device not found: Error { code: HRESULT(0x00000000), message: \"Der Vorgang wurde erfolgreich beendet.\" }"`

I'm happy to help, just tell me what you need.

Oppzippy, 2024-12-23:
Thanks, I have all the information I need from another similar issue. I'm finishing up with #101 first, since the remaining issues for that actually affect all devices. After that, I'll start on this.

Oppzippy, 2025-01-08:
As of v1.18.0, bluetooth classic is used rather than bluetooth low energy, so that should fix the problem. Please let me know if it works now.

flue17, 2025-01-08:
I tested in the Android app. Everything works except for "Quick Presets":
[image]

Thanks for the fix!

Oppzippy, 2025-01-10:
v1.18.1 should fix quick presets

flue17, 2025-01-10:
They're working now. Thank you

platypusgit, 2025-01-10:
It works, thank you very much.

closed 2025-01-10 by Oppzippy