Oppzippy/OpenSCQ30 #144: Now working on Debian 12 with soundcore Life Q30

tbartold, 2024-12-27:
Device is registered:

~/SoundcoreDesktop$ bluetoothctl
Agent registered
[bluetooth]# devices
Device 98:47:44:A4:E4:DC soundcore Life Q30

But application does not find it:

~/SoundcoreDesktop$ python3 SoundcoreDesktop.py 
[]
Traceback (most recent call last):
  File "/home/bartold/SoundcoreDesktop/SoundcoreDesktop.py", line 62, in <module>
    Headphone = Soundcore(macaddress, onHeadphoneEvent)
                          ^^^^^^^^^^
NameError: name 'macaddress' is not defined

If I add:   macaddress = "98:47:44:A4:E4:DC" it's not finding it:

~/SoundcoreDesktop$ python3 SoundcoreDesktop.py 
[]
98:47:44:A4:E4:DC
Traceback (most recent call last):
  File "/home/bartold/SoundcoreDesktop/SoundcoreDesktop.py", line 70, in <module>
    Headphone.connect()
  File "/home/bartold/SoundcoreDesktop/SoundcoreAPI.py", line 196, in connect
    raise ConnectionError(f"Cannot connect to {self.macaddress}")
ConnectionError: Cannot connect to 98:47:44:A4:E4:DC

If I now change line 67 in the API to "soundcore" (lowercase) I now get another problem:

~/SoundcoreDesktop$ python3 SoundcoreDesktop.py 
98:47:44:A4:E4:DC
Traceback (most recent call last):
  File "/home/bartold/SoundcoreDesktop/SoundcoreDesktop.py", line 70, in <module>
    Headphone.connect()
  File "/home/bartold/SoundcoreDesktop/SoundcoreAPI.py", line 192, in connect
    self.client.connect((self.macaddress, self.Port))
OSError: [Errno 16] Device or resource busy

I don't know what to do next, but I think I might be bypassing something by forcing the macaddress?

Oppzippy, 2024-12-27:
I assume you probably meant to submit the issue here: https://github.com/KillerBOSS2019/SoundcoreDesktop/issues
This project is not SoundcoreDesktop, and none of it is written in Python.

tbartold, 2024-12-28:
Sorry. Thanks for the redirect.

closed 2024-12-28 by tbartold