Oppzippy/OpenSCQ30 #147: Support Dot 3i

musaoruc, 2025-01-11:
Support for Dot 3i earphones would have been a nice addition. Thank you in advance.

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

musaoruc, 2026-01-09:
`Some(Packet { command: Command([1, 1]), body: [1, 1, 5, 5, 0, 0, 49, 50, 46, 48, 53, 49, 50, 46, 48, 53, 51, 57, 56, 50, 57, 53, 102, 101, 54, 49, 97, 52, 53, 54, 53, 99, 15, 0, 110, 130, 150, 150, 130, 110, 100, 90, 120, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 98, 1, 86, 1, 99, 1, 84, 1, 1, 1, 0, 3, 0, 0, 1, 0, 255, 255, 1, 0, 0, 0, 0, 49, 0, 0, 0, 0] })`

Here is the required information for this device.