Oppzippy/OpenSCQ30 #149: Support for Soundcore R50i NC

fzzzn, 2025-01-14:
Hello Id like to add my device to the request for support. Thanks.

```
 2025-01-14T16:04:23.944077Z ERROR  select device

Caused by:
    0: D:\a\OpenSCQ30\OpenSCQ30\lib\src\soundcore_device\device\soundcore_device.rs:86:41: TryIntoInboundPacketError { message: "Error(VerboseError { errors: [([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Nom(MapOpt)), ([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context(\"left double click\")), ([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context(\"custom button model\")), ([1, 1, 5, 6, 255, 255, 48, 49, 46, 54, 52, 48, 49, 46, 54, 52, 51, 57, 53, 57, 68, 69, 68, 54, 54, 57, 50, 68, 66, 54, 70, 52, 254, 254, 101, 120, 161, 171, 171, 152, 144, 179, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context(\"a3945 state update packet\")), ([1, 1, 5, 6, 255, 255, 48, 49, 46, 54, 52, 48, 49, 46, 54, 52, 51, 57, 53, 57, 68, 69, 68, 54, 54, 57, 50, 68, 66, 54, 70, 52, 254, 254, 101, 120, 161, 171, 171, 152, 144, 179, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Nom(Alt))] })" }
    1: Error(VerboseError { errors: [([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Nom(MapOpt)), ([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context("left double click")), ([10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context("custom button model")), ([1, 1, 5, 6, 255, 255, 48, 49, 46, 54, 52, 48, 49, 46, 54, 52, 51, 57, 53, 57, 68, 69, 68, 54, 54, 57, 50, 68, 66, 54, 70, 52, 254, 254, 101, 120, 161, 171, 171, 152, 144, 179, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Context("a3945 state update packet")), ([1, 1, 5, 6, 255, 255, 48, 49, 46, 54, 52, 48, 49, 46, 54, 52, 51, 57, 53, 57, 68, 69, 68, 54, 54, 57, 50, 68, 66, 54, 70, 52, 254, 254, 101, 120, 161, 171, 171, 152, 144, 179, 120, 120, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 10, 241, 240, 102, 102, 242, 243, 68, 68, 51, 0, 85, 0, 0, 1, 255, 1, 49, 1, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], Nom(Alt))] })
    at gui\src\main.rs:288

  2025-01-14T16:04:23.946293Z  WARN  inbound_packets_channel: error receiving packet: Other { source: Error { code: HRESULT(0x800703E3), message: "The I/O operation has been aborted because of either a thread exit or an application request." }, location: Location { file: "D:\\a\\OpenSCQ30\\OpenSCQ30\\lib\\src\\soundcore_device\\connection\\windows\\windows_connection.rs", line: 237, col: 21 } }
    at lib\src\soundcore_device\connection\windows\windows_connection.rs:258
```

kRHYME7, 2025-09-16:
Just recently got the device so I'll subscribing to this thread. 

Build from master branch.

[image]

kRHYME7, 2025-09-20:
@Oppzippy  It now proceeds. I wanted to help test this. Any pointers on what should I start tinkering?

Oppzippy, 2025-09-20:
Configuring the buttons is the main thing I'm not sure if I have right. So if you can make sure that when changing each button in the UI, the change is properly reflected on the device, that would be very helpful.

kRHYME7, 2025-09-21:
> Configuring the buttons is the main thing I'm not sure if I have right. So if you can make sure that when changing each button action in the UI, the change is properly reflected on the device, that would be very helpful.

I will try it then. Last time I check almost all buttons are not working. Any way to have verbosed logs or stdout of the `cargo run --bin openscq30-gui     ` is enough?

Oppzippy, 2025-09-22:
Please try the latest commit and let me know if that fixes button configuration.

`RUST_LOG=none,openscq30_lib=debug,openscq30_gui=debug cargo run --bin openscq30-gui` will show debug logs for openscq30 and hide logs for its dependencies, since `wgpu` debug logs are very noisy. In this case, I know what's wrong, though. I knew for sure that the R50i NC accepted a command for setting the action of each button individually, but I wasn't sure if it accepted the command that sets all buttons' actions together in bulk, which is the one I was using. It seems it does not accept the bulk set command, so I switched to the singular repeated for each changed button.

Oppzippy, 2025-09-24:
This turned out to be a bug with openscq30's bluez backend, so I was able to reproduce this with a different device. Fixed in bb39a82d0b8d79a669ce798732b24785e42def41.

As a side note, the missing icons (window close button, dropdown down arrow, etc) are due to not having [cosmic-icons](https://github.com/pop-os/cosmic-icons/) installed. I hadn't mentioned it in the documentation until a few days ago. Distros that package it call it either `cosmic-icons` or `cosmic-icon-theme`.

kRHYME7, 2025-09-29:
Reporting...

 EQ now works. Thank you very much!!! 

In `Sound Modes` nothing changed, still borke.

kRHYME7, 2025-10-01:
@Oppzippy  It now works! Thank you! Looks like it is ready for daily use.

closed 2025-10-29 by Oppzippy