Oppzippy/OpenSCQ30 #152: Error: trying to connect to Space A40

TeoNero, 2025-01-23:
Good afternoon, 

first of all I wanted to thank you so much for this work, since I've been searching for a solution like this for a long time. I must say in advance that I have little to no technical knowledge, so I may be the issue here.  But I have Space A40 headphones that the software correctly recognizes (see photo), but I am unable to finalize the connection and get into the software.

 Console gives me this error:

 2025-01-23T16:25:11.370386Z  WARN  inbound_packets_channel: error receiving packet: Other { source: Error { code: HRESULT(0x800703E3), message: "Operazione di I/O terminata a causa dell'uscita dal thread oppure della richiesta di un'applicazione." }, location: Location { file: "D:\\a\\OpenSCQ30\\OpenSCQ30\\lib\\src\\soundcore_device\\connection\\windows\\windows_connection.rs", line: 237, col: 21 } }
    at lib\src\soundcore_device\connection\windows\windows_connection.rs:258

remaining: 125
remaining: 125
remaining: 125
  2025-01-23T16:25:11.492047Z ERROR  select device

Caused by:
    missing device serial number
    at gui\src\main.rs:288

Thank you so much for you work and help!
Matteo

[image]

RodoMa92, 2025-03-02:
Try again with the latest version, there is another thread [here](https://github.com/Oppzippy/OpenSCQ30/issues/101#issuecomment-2631246500) where a lot of progress has been made on this, now it should be far more usable :)

RodoMa92, 2025-12-24:
@Oppzippy Just tested with 2.0.1 and I can confirm that now everything looks correct and works fine (besides a couple of visual gliches in the UI, but that's not really a problem with your code, it's probably just my desktop compositor/UI rendering issues).

Oh, just the serial number looks incorrect, but it's really a nitpick TBH XD

This can be closed. Thanks a ton for your work! :)

closed 2025-12-25 by Oppzippy