Oppzippy/OpenSCQ30 #154: Fit to screen width (Mobile)

misc-de, 2025-01-25:
Hi !

I tried to use the app on my FLX1 mobile device, but it doesn't fit to the screen size - take a look on the picture :)

[image]

I hope, it is possible to fix the UI to be more flexible.

Oppzippy, 2025-01-25:
Interesting, I hadn't considered Linux phones, so I didn't bother to make the UI responsive. I have been considering rewriting the GUI in [libcosmic](https://github.com/pop-os/libcosmic) for a while, and I'm currently in the process of making large architectural changes to the application, which gives a good opportunity for a UI rewrite. I'm probably going to put this off until then.

misc-de, 2025-01-26:
Linux phones have become popular over the last 5 years. In the last 12 months in particular, things have picked up a bit of momentum and are now usable for daily use. Some of the applications are not yet adapted to the screens, but this is getting better and better.

If you make the effort, even if it takes a while, we will all have gained something. Thank you very much for your work and this app :)

pinoverclock, 2025-02-08:
@misc-de Looks like you are using the pinephone just like me, I'm using the pinephone with archlinux.
I have overcomed that problem by using this command:
```
scale-to-fit com.oppzippy.OpenSCQ30 true
```
If you you don't have that script, here is the scale-to-fit code:
```
#!/bin/bash

set -e

export GSETTINGS_SCHEMA_DIR=_build/data/

function print_current()
{
  local app_id="$1"
  local munged_id="$2"

  echo -n "Scale-to-fit for $app_id: "
  G_MESSAGES_DEBUG='' gsettings get sm.puri.phoc.application:/sm/puri/phoc/application/"$munged_id"/ scale-to-fit
}

if [ -z "$1" ]; then
  echo "Usage: $0 APP-ID [VALUE]"
  exit 0
fi

APP_ID="$1"
MUNGED_ID=$(echo "$1" | sed 's/[^a-zA-Z0-9]/-/g' | tr '[:upper:]' '[:lower:]')

if [ -z "$2" ]; then
  print_current "$APP_ID" "$MUNGED_ID"
  exit 0
fi

case "$2" in
	""|on|1|ON|true)
	val=true
	;;
	*)
	val=false
	;;
esac
G_MESSAGES_DEBUG='' gsettings set sm.puri.phoc.application:/sm/puri/phoc/application/"$MUNGED_ID"/ scale-to-fit "$val"

print_current "$APP_ID" "$MUNGED_ID"

```

I hope that helps :)

misc-de, 2025-02-08:
@pinoverclock Thank you for the script.

When I read "scale-to-fit" I remembered that I have the mobile settings app with this function. Unfortunately I have a 225% ratio on the display and the fonts are very small. But it is definitely easier to use this way.

Your script fulfills exactly the same function as the app but I like your clean script :)

pinoverclock, 2025-02-09:
@misc-de Thank you, but that script it's not mine but community's. I guess that app you are talking about enables the global property scale-to-fit ```gsettings set sm.puri.phoc scale-to-fit true```
With the scale-to-fix script you can set individually on which apps you the scale-to-fix property enabled or not.

Anyway this is just a way around method to be able to work with apps that a not able to resize themselves properly, the correct way to go to make the apps self resizable so they can adapt to mobile's smaller screens.
And yes, same problem here, the fonts gets really tiny when you scale the app this way, and I'm only using a 150% ratio on the display.
Looks like the FLX1 is an overall better device than the pinephone or the pinephone pro, I will keep an eye on the evolution of this phone.