Oppzippy/OpenSCQ30 #155: Better Android integration

gro-david, 2025-01-25:
Is there a way of integrating the app a bit better with android? For me material you, and widget support would come to mind first.

Love the app otherwise <3

KraXen72, 2025-12-05:
Hello, it would be great if the app has a widget for toggling between ANC / Normal / Transparency. 
I have my space a40 set up to only toggle between normal & ANC, since i use transparency only ocassionally. A widget would make this much quicker to do, though.
Thanks!

Oppzippy, 2026-01-15:
Material you has always been enabled by default on android versions that support it. [v2.3.0](https://github.com/Oppzippy/OpenSCQ30/releases/tag/v2.3.0) adds a widget that displays a configurable list of settings.

Oppzippy, 2026-01-16:
If you mean [IzzyOnDroid's repo](https://apt.izzysoft.de/fdroid/index/apk/com.oppzippy.openscq30), I think he said he's going to update openscq30 monthly, so expect it to be available around February 1. If you mean [https://f-droid.org/](https://f-droid.org/), it would be nice to have openscq30 available there at some point, but it's not really a priority for me right now since IzzyOnDroid also verifies reproducible builds.