Oppzippy/OpenSCQ30 #163: Request for soundcore Liberty 4 NC

denis-robel, 2025-05-22:
is it possible to support the soundcore Liberty 4 NC?

It would be great to have it because of I want to run OpenSCQ30 on my linux phone.

best regards Denis

rzayoruk, 2025-07-12:
First of all, thank you to the developers for this amazing project! I am experiencing the exact same issue with my soundcore liberty 4 NC headphones. I'm happy to provide any additional logs, system information, or help with testing if needed. Hoping for a fix and looking forward to being able to use this great tool!
Best regards.

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

Oppzippy, 2025-10-31:
Closing in favor of an older request for the same device