Oppzippy/OpenSCQ30 #164: Request for Soundcore P30i

ArfusUwU, 2025-06-14:
Per the title. 

There's a similar project for nothing earbuds, I'd love this for my anker devices.

victay44, 2025-07-25:
See [this](https://github.com/Oppzippy/OpenSCQ30/issues/149)

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

Oppzippy, 2025-10-31:
Actually, it looks like this has the same model number as the R50i NC, so it may be the same thing with different branding. Please try using v2.0.0-beta4 (or newer) and connecting to your P30i while selecting Soundcore R50i NC as the device model.

If that doesn't work, then see my previous message.