Oppzippy/OpenSCQ30 #169: Report for Liberty 4 Pro A3954 (A3954GG1)

thymaro, 2025-07-01:
The app shows the headphones on initial startup of the app, pairing is possible upon tapping the record in the list. After initial pairing, tapping the model in the list, a new screen appears with "loading" for ca. 3 seconds. It quickly goes to "disconnected" for ca. 200 ms, then back to the list. This is all.

If you manage to support the model, wave my way and I will test.

EldrinSMP, 2025-10-21:
I second for the testing.

Oppzippy, 2025-10-31:
To anyone that has this device, please see https://github.com/Oppzippy/OpenSCQ30/blob/master/docs/request-soundcore-device.md and post the information

Oppzippy, 2025-10-31:
@AlirezaPourchali 
> do i need to have a specific sqlite3 version?

Theoretically it should work with sqlite 3.45.1 or newer, since that's what ubuntu 24.04 is using and the tests pass there. I'll look into this.

thymaro, 2025-10-31:
Sorry, no, it was my mistake. I should have gone step by step instead of leaping and leaving out steps. I am so masculine!

Here's the requested information.

`Some(Packet { command: Command([1, 1]), body: [0, 1, 99, 100, 0, 0, 48, 51, 46, 50, 57, 48, 51, 46, 50, 57, 51, 57, 53, 52, 68, 69, 55, 55, 53, 49, 56, 65, 57, 68, 70, 52, 48, 50, 46, 53, 56, 9, 244, 157, 138, 83, 43, 186, 254, 254, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 0, 1, 145, 147, 139, 141, 122, 111, 105, 100, 60, 60, 145, 147, 139, 141, 122, 111, 105, 100, 60, 60, 0, 0, 0, 0, 1, 145, 147, 139, 130, 122, 133, 133, 114, 60, 60, 145, 147, 139, 130, 122, 133, 133, 114, 60, 60, 0, 0, 10, 102, 102, 50, 51, 255, 255, 68, 68, 51, 2, 6, 0, 0, 255, 1, 255, 0, 0, 0, 0, 94, 1, 110, 1, 1, 0, 0, 1, 0, 1, 95, 0, 0, 1, 1, 0, 0, 0, 110, 1, 0, 1, 0, 0, 255, 0, 0, 17, 17] })`

EldrinSMP, 2026-06-15:
Hi! Just wondering if there was any updates on the timeline for implementing these or if there was a anything us owners might can be able to do it help!

closed 2026-06-29 by Oppzippy

Y-PLONI, 2026-09-23:
@Oppzippy 
I’m having the same issue described here:
https://github.com/Oppzippy/OpenSCQ30/issues/246
I downloaded this file:
https://github.com/Oppzippy/OpenSCQ30/releases/download/v2.11.0/openscq30-gui-installer-windows-x86_64.exe
What should I do?
Thanks a lot for the great software!
Would you be interested in a PR that adds a light mode and rounded styling for Windows?

[image]